esp8266/Arduino #3021: Arduino vs Espressif Flash download Tools - Are they flashing all the necessary bin files? | WiFi Dropouts

9lash, 2017-03-04:
Hello All: 

I built a custom board which has ESP8266 and a 4Mbit flash (means 512KB ) in it. I am experiencing WiFi drop out issues. I learnt that SDK 2.0 upgrade solves this problem in a couple of posts in git.  I read through both the routes of programming an ESP8266 through an Arduino way and as well as the Espressif way using Flash download tools. 

In an example in SDK 2.0, which describes how to flash a program in ESP8266 using the Flash download tools, they mention that I need to flash the following bin files at specific locations: 

Non OTA firmware Map
master_device.bin ---> 0x3E000
esp_init_data_default.bin --> 0x7C000
blank.bin ---> 0x7E000
boot
user2.bin --> 0x0000      (This bin file looks like a combination of eagle.flash.bin + eagle.irom0text.bin + the program that user is implementing in the ESP8266 ) 

When I program my code using an Arduino, I dont see the other bin files like esp_init_data_default and boot going into the Flash. **Does Arduino flash all the other bin files?**  

I am concerned if that is the reason I am getting WiFi drops outs. 

To solve this problem, Lets say I have written my program in Arduino environment (helloserver.ino). When I compile this file, Arduino makes  a helloserver.ino.bin file in a temp folder. Lets say I grab that file and place it in flash downloader tool as the user2.bin file above. And place all the other bin files as mentioned above in the correct locations. Will this meet the requirements mentioned in the SDK documentation? 

Please let me know: 
**1. How to configure SDK 2.0 branch to work in Arduino IDE?** 
**2. How to configure rf_cal_sector_set in Arduino IDE?** 

### Basic Infos

#### Hardware
Hardware:		ESP-01 look alike
Core Version:      	2.3.0

### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512KB/No SPIFFS
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		ck 

### Sketch

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

 
// WiFi parameters
const char* ssid = "DGA_front";
const char* password = "password";

ESP8266WebServer server(80);

const int led = 13;

void handleRoot() {
  digitalWrite(led, 1);
  server.send(200, "text/plain", "hello from esp8266!");
  digitalWrite(led, 0);
}

void handleNotFound(){
  digitalWrite(led, 1);
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
  digitalWrite(led, 0);
}

void setup(void){
  pinMode(led, OUTPUT);
  digitalWrite(led, 0);
  Serial.begin(115200);
  WiFi.disconnect();
  delay(100);
  
  WiFi.mode(WIFI_STA);
  delay(100);
  
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.print("PHY Mode: ");
  Serial.println(WiFi.getPhyMode());

  Serial.println(" PHY_MODE_11B = 1, PHY_MODE_11G = 2,PHY_MODE_11N = 3");
 
 /* 
  Serial.println("Setting it to 2");
  //Setting the Physical layer mode in 801
  WiFi.setPhyMode(WiFiPhyMode_t(1));
  Serial.print("New PHY Mode: ");
  Serial.println(WiFi.getPhyMode());
*/
  
  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });

  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");
  
}

void loop(void){
  server.handleClient();
}
```

### Debug Messages

```
wifi evt: 1
STA disconnect: 2
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
....wifi evt: 1
STA disconnect: 2
....wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
..wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
.wifi evt: 1
STA disconnect: 4
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.....wifi evt: 1
STA disconnect: 2
.....wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
.wifi evt: 1
STA disconnect: 2
.......wifi evt: 1
STA disconnect: 4
wifi evt: 7
.wifi evt: 7
wifi evt: 7
```

Shumatic, 2017-06-19:
Hi
I can't answer you particular question, but maybe this will help.

void loop(void){
  server.handleClient();
  yield()  // This gives the CPU time to do background stuff
}

devyte, 2017-09-07:
This is documented. I can vouch there being no current issues with the nodemcu module.
In any case, this is not the right place to ask about issues with your board. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for assistance. If after discussion there you find a reproducable issue in the code in this repo, please open a new issue with minimal and complete details.
This looks like either user error or hardware defect. Closing.

closed 2017-09-07 by devyte

Shumatic, 2017-09-19:
just trying to help