esp8266/Arduino #3023: ISSUE :- error: espcomm_open failed in NODEMCU 8266 V0.9 in arduino 1.8.1

brijeshyati, 2017-03-05:
Dear Team ,

I am beginner in arduino (version 1.8.1 , ESP 8266 NODEMCU DEVKIT V0.9 ). 
currently i am facing issues . need you guy help in it 
for  first couple of program ( just basis one ) blinking of LED etc,it is working fine ( ESP 8266 ). I am able to see blinking of LED bulb at ESP8266 . 
But suddenly it stopped and after that whenever trying to compile any  program. getting below error 
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

I fed up with this errorst . 

Regards,
Brijesh Kumar

jjssoftware, 2017-03-06:
https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/faq/a01-espcomm_sync-failed.md

harshadgd, 2017-06-23:
Hi, I had exactly same issue. The problem was resolved by folllowing these steps
1. Connect Vin of the board to 6V ( you can use 4 AA batteries or you can use 9V battery)
2. Connect the GND pin which is next to Vin Pin to ground (-ve on breadboard)
3. Connect D3 pin of the board to GND which will force your ESP8266 to flash mode.
4. On Arduino Ide application use 'ESP generic board'. Node mcu 12e board selection will not work.
5. Make sure the flash mode option is set to DIO under tools. This is the defaul option.
6. Try compiling and uploading the example program BLINK.
7. The blue LED will start flashing during the upload and I am pretty sure the upload will be successful.

Pls reply if this doesn't work.

ericweintraub-qsft, 2017-07-02:
I had this problem and wasted two days trying to fix. I happened to have bought two of them and sure enough the other one works fine. I think this simply may be a hardware defect. Thanks to Amazon prime, will simply return for another.

Eric2XU, 2017-07-02:
I had this problem and wasted two days trying to fix. I happened to have bought two of them and sure enough the other one works fine. I think this simply may be a hardware defect. Thanks to Amazon prime, will simply return for another.

devyte, 2017-09-09:
@brijeshyati the message is indicative of not being able to open the serial port. Either you have (had?) an installation problem of your flashing system or serial port drivers, the serial port id changed (i.e.: you unplugged/replugged the module), or your module is dead (I saw such a case just today myself).
In any of the above, this is not the right place to ask for usage assistance. Please refer to the docs, or ask at esp8266.com or stackoverflow for help.
I can vouch that the flashing system behaves correctly in the latest git today, at least under windows and linux. Closing this.

closed 2017-09-09 by devyte

shivamkeshri, 2017-12-04:
If you get error message like this:
Archiving built core (caching) in: C:\Users\skeshri\AppData\Local\Temp\arduino_cache_820988\core\core_esp8266_esp8266_nodemcuv2_CpuFrequency_80,UploadSpeed_115200,FlashSize_4M3M_b4034d01e7e0e33ff4df941b74132332.a
Sketch uses 222205 bytes (21%) of program storage space. Maximum is 1044464 bytes.
Global variables use 31572 bytes (38%) of dynamic memory, leaving 50348 bytes for local variables. Maximum is 81920 bytes.
Uploading 226352 bytes from C:\Users\skeshri\AppData\Local\Temp\arduino_build_170968/Blink.ino.bin to flash at 0x00000000
error: failed sending 1072 bytes
error: failed sending 0xC0
warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 4 bytes
error: failed sending 0xC0
warning: espcomm_send_command: didn't receive command response
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

Then your port is not supplying enough current to boot your program for this you change the USB port.

arif8051, 2018-02-09:
Hello friends.....
need help....
I can't upload program node mcu lolin v3
compile no problem
but upload time show error message.....
advance thanks friends....
error message is-
Arduino: 1.8.6 Hourly Build 2018/01/03 03:33
(Windows 10), Board: "NodeMCU 1.0 (ESP-12E
Module), 80 MHz, 9600, 4M (3M SPIFFS)"
Archiving built core (caching) in: C:\Users
\MICROS~1\AppData\Local\Temp\arduino_
cache_943422\core\core_esp8266
_esp8266_nodemcuv2_CpuFrequenc
y_80,UploadSpeed_9600,FlashSiz
e_4M3M_9cfa2c1632441f3f34830c9
cc835a77d.a
Sketch uses 222209 bytes (21%) of program
storage space. Maximum is 1044464 bytes.
Global variables use 31576 bytes (38%) of
dynamic memory, leaving 50344 bytes for local
variables. Maximum is 81920 bytes.
Uploading 226352 bytes from C:\Users\MICROS
~1\AppData\Local\Temp\arduino_
build_596374/Blink.ino.bin to flash at
0x00000000
error: failed sending 1072 bytes
.error: failed sending 1044 bytes
.error: failed sending 1040 bytes
.error: failed sending 1040 bytes
.error: failed sending 1070 bytes
.error: failed sending 1083 bytes
.error: failed sending 1069 bytes
.error: failed sending 1080 bytes
warning: espcomm_send_command: didn't
receive command response
warning: espcomm_send_command(FLASH_DOW
NLOAD_DATA) failed
warning: espcomm_send_command: wrong
direction/command: 0x01 0x03, expected 0x01
0x04
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed
This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Pablo2048, 2018-02-09:
Set serial speed at least to 115200 (and next time do some search on the internet and/or post this kind of questions to the forum....)

arif8051, 2018-02-09:
brother lolin v3 reated buad speed is 9600.....

I search lot of time,  read many articles and watch lot of tutorial but error message show.......

Pablo2048, 2018-02-09:
OMG, brother, its only printed on PCB and is valid for Lua interpreter, not for Arduino (I'm pretty sure that I already mentioned this somewhere in the forum...)

lrmoreno007, 2018-02-09:
https://github.com/esp8266/Arduino/blob/master/doc/faq/a01-espcomm_sync-failed.rst

arif8051, 2018-02-09:
so what can I do for upload program.......
please help brother.....

Pablo2048, 2018-02-09:
Do what I wrote - set correct speed. If this fails, play with reset mode...

lrmoreno007, 2018-02-09:
You have Arduino: 1.8.6 Hourly Build, test with Arduino 1.8.5 which is the actual stable version.

arif8051, 2018-02-10:
Thanks  to all brother.....
Done uploading....

Tifa80, 2018-02-25:
Had the same problem on one sketch and the other was working fine
Couldn't find the difference between the setting of the two sketches
copied the code from the problematic sketch to the other one and it worked fine

iCounterBOX, 2018-03-12:
Hi
I had the same failure this weekend ( same issue during code load out of Eclipse Sloeber )
I read the hint from above from **harshadgd** ....but to be honest i did not try...It is now my 10 th  NodeMCU which i use here...And the FIRST time with this issue.
This issue was just shown from 1 MCU from a set of 6 which i ordered.

concerning code:
Cannot imagine that this "code-related"..we use this code 6 MCU´s all the time..all same...the issue was only with this 1 MCU.

Today i got the answer from my dealer! They have from time to time such issues..And for sure he will send me a NEW MCU. So if this device is unter warrianty i would recommend to send it back..

regards
K

baileyt3, 2018-03-28:
Hi,
I have the same issue of errors and failure to upload, nothing seemed to fix it, I was using Arduino IDE, I had the occasional success, but mostly failures, I spent many hours with no success of fixing it. 
My connection to the ESP-12 was via USB, I also had a serial RX/TX lead connected to my Pi using Minicom to see if I could help debug the problem. Normally I just get garbage on the Minicom screen when I boot the ESP.
I also tried sending 'AT' commands via this serial connection, but noticed no command ever worked.
I had an idea that the TX line from the Pi to the RX on the ESP-12 might be an issue, so I removed it.
BINGO it all started working, I can now upload with no problems, 
It would appear there is some sort of conflict if you have USB -> Pi and serial to TX/RX that prevents it from working. 
I hope this may help others, basically don't connect anything to the RX (GPIO 3) on the ESP-12 when uploading.

pmerlinojr, 2018-03-29:
This definitely works:

posed by harshadgd commented on Jun 22 2017  
Hi, I had exactly same issue. The problem was resolved by following these steps
Connect Vin of the board to 6V ( you can use 2 AA batteries or you can use 9V battery)
Connect the GND pin which is next to Vin Pin to ground (-ve on breadboard)
Connect D3 pin of the board to GND which will force your ESP8266 to flash mode.
On Arduino Ide application use 'ESP generic board'. Node mcu 12e board selection will not work.
Make sure the flash mode option is set to DIO under tools. This is the defaul option.
Try compiling and uploading the example program BLINK.
The blue LED will start flashing during the upload and I am pretty sure the upload will be successful.

heynomansheikh, 2018-05-29:
just unplug 3V3 pin before uploading, it will work.

shinuza, 2018-05-31:
I have Lolin NodeMCU V3, I followed, [these instructions](http://www.instructables.com/id/Getting-Started-With-ESP8266LiLon-NodeMCU-V3Flashi/) and to upload use NodeMCU firmware programmer, you have to hold down the flash button and then press the reset button. That's the only way I go the NodeMCU Firmware programmer to work.

From there I was able to upload a sketch.

Pradeep-1612, 2018-06-12:
I'm not able to proceed further becoz of an error :
"Board nodemcv2 (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 1.0 (ESP-12E Module).  "
How can fix this error ?
Plzz.... Reply Ur Solutions Quick !!
Thank Ü !

baileyt3, 2018-06-12:
Pradeep-162, we would need to know more about your situation, what are you trying to compile, what IDE are you using, what platform (Windows or Unis (Pi)).
One thing to check though is have you selected the correct COM port. Tools->Port setting

Pradeep-1612, 2018-06-12:
Hi,
I'm using Arduino IDE.
I'm trying to compile secret room protector program.
I'm using WINDOWS.
I'm selected COM4.

Pradeep-1612, 2018-06-13:
I'm not able to proceed further becoz of an error :
"Board nodemcv2 (platform esp8266, package esp8266) is unknown

Error compiling for board NodeMCU 1.0 (ESP-12E Module). "

I'm using Arduino IDE.
I'm trying to compile secret room protector program , and it's showing the same error for all programs.
I'm using WINDOWS 10.
I'm selected COM4.

How can fix this error ?
Plzz....Reply Quick .
Thank Ü !

lrmoreno007, 2018-06-13:
Read de FAQ in the documentation.

romelemperado, 2018-08-06:
I have encountered this issue too..  In my case There is a short circuit between the clk and grnd pins..

clubbyr8, 2018-09-01:
[image]
Similar issue, a solder bridge between RST and GND pins. Works perfectly now...

romelemperado, 2018-09-01:
it looks like they soldered it manually.. :)

adavat, 2018-09-30:
I had this issue with a ESP8266 Generic on which I already managed to upload. It worked when I changed my USB-cable for another one. In both cases, the light flashes on connexion, but with the good cable a new COM port appear in the list.

Zebz13, 2019-01-17:
> Hi, I had exactly same issue. The problem was resolved by folllowing these steps
>
> 1. Connect Vin of the board to 6V ( you can use 4 AA batteries or you can use 9V battery)
> 2. Connect the GND pin [...]

It did not work for me. I even tried flashing firmware. Still didn't work. Never got a response from ESP.
The inbuilt LED blinks only when I connect it to usb port or when I press the reset button.

AbeeB, 2019-03-24:
Iam currently using arduino 1.8.9v .When i atempted to upload simple examples to my nodemcu.,the following error is shown ....
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

please could anyone sort it out.........

pmerlinojr, 2019-03-25:
I get that often too.I close the Arduino API and reboot and it usually fixes it.

Also make sure you have the correct board and Port selected.

Sent from XFINITY Connect App

------ Original Message ------

From: AbeeB
To: esp8266/Arduino
Cc: Paul Merlino, Comment
Sent: March 24, 2019 at 2:03 PM
Subject: Re: [esp8266/Arduino] ISSUE :- error: espcomm_open failed in NODEMCU 8266 V0.9 in arduino 1.8.1 (#3023)

Iam currently using arduino 1.8.9v .When i atempted to upload simple examples to my nodemcu.,the following error is shown .... warning: espcomm_sync failed error: espcomm_open failed error: espcomm_upload_mem failed error: espcomm_upload_mem failed
please could anyone sort it out.........

AbeeB, 2019-03-25:
Thanku so much sir ...in addition to what you have said ,clearing the temp
files and the old versions it have solved my issue

ankit12345678, 2019-05-11:
Hi,
i am using esp8266 12E NODE MCU1.0, with 1.8.5 version of ARDUINO IDE , i have installed all the drivers related to esp8266 12e but getting the (espcomm_upload_mem failed error)when i am uploading simple blink code every time,

Please help me.

pmerlinojr, 2019-05-11:
Close Arduino. Clear all temp tables from your stored files.

Reboot and you should be fine.

Sent from XFINITY Connect App

------ Original Message ------

From: ankit12345678
To: esp8266/Arduino
Cc: pmerlinojr, Comment
Sent: May 11, 2019 at 12:02 PM
Subject: Re: [esp8266/Arduino] ISSUE :- error: espcomm_open failed in NODEMCU
8266 V0.9 in arduino 1.8.1 (#3023)

Hi,
i am using esp8266 12E NODE MCU1.0, with 1.8.5 version of ARDUINO IDE , i have installed all the drivers related to esp8266 12e but getting the (espcomm_upload_mem failed error)when i am uploading simple blink code every time,

Please help me.

kishorebolt03, 2019-08-23:
> just unplug 3V3 pin before uploading, it will work.

thank you bhai