esp8266/Arduino #3024: Mac address change with initVariant() lost after ESP.restart()

torntrousers, 2017-03-05:
I need to change the AP mac address so use the approach from Issue 1221:

`extern "C" {
  #include "user_interface.h"
}
void initVariant() {
  uint8_t mac[] = {0x36, 0x12, 0x34, 0x56, 0x78, 0xAB};
  wifi_set_macaddr(SOFTAP_IF, &mac[0]);
}
`
And that works fine, however then doing an ESP.restart() and that mac change is forgotten and it reverts back to the default mac address.

Is there some way to get that initVariant() code to run on the restart too?

(I'm using the latest Github code)

devyte, 2017-09-07:
@torntrousers did you test setting with the following:
WiFi.macAddress(mac);
Wifi.softAPmacAddress(mac);
?

devyte, 2017-10-10:
No feedback in over a month, closing.

closed 2017-10-10 by devyte

torntrousers, 2018-04-09:
This is still broken on the latest code in git. The comment above about  WiFi.macAddress(mac);
Wifi.softAPmacAddress(mac); is irrelevant as they get the mac not set it. See https://github.com/HarringayMakerSpace/ESP-Now/issues/5#issuecomment-379916598 but to sumarize that - a work around is to erase the entire flash and then include in the sketch a call to WiFi.persistent(false).

devyte, 2018-04-09:
The mac setter/getters in the core are a thin wrapper around Espressif's SDK functions wifi_s|get_macaddr(). From the SDK doc 2c esp8266 nonos sdk api reference, these functions are supposed to be called from user_init() on bootup, and there is no mention of storing the values in flash along with the wifi credentials et al.
In other words, we're already abusing the sdk functionality by allowing to change the mac at runtime.

torntrousers, 2018-04-10:
There is clearly a bug somewhere here though. Would you reopen this and I can put give a simple sketch that shows the problem.

devyte, 2018-04-10:
Open a new issue, fill everything out, include the sketch and explanation, and I'll take a look.

torntrousers, 2018-04-10:
Writing the minimal sketch for you I found out more what the issue is. It still seems like a bug to me but now I understand I wont bother raising a new issue. I will explain here for posterity...

The reason is that wifi_set_macaddr in initVariant only seems to work correctly setting the mac on the interface if the Wifi mode is using that interface. As initVariant  runs before setup() the wifi mode when initVariant runs is the previously used mode. So if a sketch sets the mode to WIFI_STA then the ESP restarts the mode will still be WIFI_STA when initVariant runs so wifi_set_macaddr(SOFTAP_IF, &mac[0]); doesn't appear to work and the AP MAC will remain the default.

The reason erasing the flash appears to fix it is because that sets the default wifi mode which is WIFI_AP_STA - so using both the AP and STA interfaces.

Here's and example:
```
#include <ESP8266WiFi.h>
extern "C" {
  #include "user_interface.h"
}
void initVariant() {
  uint8_t mac[] = {0x36, 0x33, 0x33, 0x33, 0x33, 0x33};
  wifi_set_macaddr(SOFTAP_IF, &mac[0]);
}

void setup() {
  Serial.begin(115200); Serial.println();
  Serial.print("This node AP mac: "); Serial.println(WiFi.softAPmacAddress());
  WiFi.mode(WIFI_STA);
  Serial.print("This node AP (STA) mac: "); Serial.println(WiFi.softAPmacAddress());
  delay(3000);
  // WiFi.mode(WIFI_AP); // <----- without this wifi_set_macaddr wont work
  ESP.restart();
}

void loop() {
}
```