esp8266/Arduino #3026: FSBrowser.ino example points to an external site to download .js

terryjmyers, 2017-03-05:
https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WebServer/examples/FSBrowser

In example above, edit.htm contained in the gz in the /data folder points to a js script hosted here:
https://cdnjs.cloudflare.com/ajax/libs/ace/1.1.9/ace.js
I recommend containing ace.js in the \data folder so that it is completely self contained and does not require an internet connection or this file to always exist on the internet.

 I've verified that downloaded the ace.js, compressing to a gz file, placing this ace.js.gz into the /data folder, and editing the edit.htm(and recompressing to gz) to simply point locally ...<script src="ace.js" type="text...works fine

me-no-dev, 2017-03-06:
ace.js pulls other files too (code coloring and some other stuff) so it's not just that file. If you add all available options as languages and so on, you will find that files are quite many and space is not that little. FSBrowser As-Is can run on 64K SPIFFS

terryjmyers, 2017-03-06:
I was able to edit, edit.htm to pull ace.js from spiffs and have it fully load and functional without Internet connection. I tested by unplugging my cable modem and connecting to the esp using my wife's computer which had never connected to it before(so nothing was cached). It couldn't have pulled any other files. Am I missing something?

jjssoftware, 2017-03-06:
sum the size of all js resources pulled down by ace.js + ace.js itself. If the total is > 64k, that's what you're missing

RudyFiero, 2017-03-06:
Just a comment. If there are many files for the ESP8266 to transfer to the browser what tends to happen is that it chokes and it will take longer and sometimes the files never get delivered. The browsers send off parallel requests but the ESP can't handle many connections. 

For my applications I want all files available from my ESP because I want my stuff to work even if there is no internet. But there are a lot of applications that need the internet to function and in those cases it would be beneficial for the load to be transferred to online resources.

FSBrowser is an example application. Anyone can change it to suit their applications. So there is no need for the example to cover all situations as it is meant as an example rather than 'the' way things need to be done.

devyte, 2017-09-07:
@terryjmyers
From Content Delivery Network, on wikipedia:
The goal is to distribute service spatially relative to end-users to provide high availability and high performance.

The whole point of CDNs is to offload hosting frameworks together with your own application, and having them freely available. In this case, @me-no-dev is trying to not force hosting the framework on which the editor depends on, on the SPIFFS. Why? because ESP-01s don't have a lot of SPIFFS to begin with.
In such scenarios, the webpage loaded from the ESP tells the browser: hey, you need to get this from elsewhere, and the browser, which is usually on your phone or PC, and therfore will have internet access, will do so.

Having said the above, I am also in a situation where having a dependency on internet access is not viable. I did something similar to what you did: I got the files hosted on CDN and put them on my ESPs SPIFFS (1MB of space is plenty).
If you open the browser console, you'll see that in some cases an attempt is still made to pull other files. Because the hosting is now done on the ESP, they fail as not found. Example: a lib for syntax highlighting.

If you can't rely on internet access, like in my case, my suggestion to you is to keep the changes you already made local, like I did. Changing the example is not acceptable for some users.

Closing.

closed 2017-09-07 by devyte