esp8266/Arduino #3028: WiFi.begin half-ignoring BSSID parameter

DoNck, 2017-03-08:
### Basic Infos

#### Hardware
Hardware:			Sparkfun Thing Dev ESP8266
Core Version:      	 2_3_0

### Description

In order to speed up connection of my ESP8266 to my AP, I was trying to provide a known channel and BSSID configuration using the overloaded function ```WiFi.begin(ssid, password, channel, bssid, connect)``` instead of ```WiFi.begin(ssid, password, channel)```.

It actually does speed up the process when using the correct BSSID parameter (~1.5s instead of ~6s), however I observed a strange behaviour of this overloaded function.

**Specifying the wrong BSSID parameter, connecting takes way longer (~16s) but does not fail (as I expected).**

@kentaylor [stated](https://github.com/esp8266/Arduino/issues/2735#issuecomment-266356859) using this overloaded function without changing ssid nor password would not do anything. Reading the [documentation of the station class](https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-class.md#begin) and the code ([ESP8266WiFiSTA.cpp](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L126) if I understand correctly), I can't see anything that would skip calling ```wifi_station_set_config(&conf)```.

Reading "ESP8266EX SDK Programming Guide" did not help to understand this behaviour either.

I'm confused ! Is it a bug or a feature ? ;-)

### Settings in IDE

Module:  			Sparkfun ESP8266 Thing Dev 
CPU Frequency:		160Mhz
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <ESP8266WiFi.h>

//uncomment and complete with your own credentials:
//#define WIFI_SSID        "***"
//#define WIFI_PASSWORD    "***"

bool b_WiFi_connected_bkp = false;
unsigned long begints;

void setup() {
  Serial.begin(115200);
  Serial.print("\n");
  //Serial.print("Core version: "); Serial.println(ESP.getCoreVersion());
  //Serial.print("SDK version: "); Serial.println(ESP.getSdkVersion());
  //
  //unsigned char ap_mac[18] = { 0x11, 0x22, 0x33, 0x44, 0x55, 0x66 }; //Put your actual AP BSSID here
  unsigned char ap_mac[18] = { 0x01, 0x23, 0x45, 0x67, 0x89, 0xAB }; //Put some bullshit BSSID
  char known_bssid[18] = { 0 };
  sprintf(known_bssid, "%02X:%02X:%02X:%02X:%02X:%02X", ap_mac[0], ap_mac[1], ap_mac[2], ap_mac[3], ap_mac[4], ap_mac[5]);
  Serial.print("---\nTrying to connect to AP with BSSID: ");
  Serial.println(known_bssid);
  //
  begints = millis();
  //WiFi.begin(WIFI_SSID, WIFI_PASSWORD); //for comparison
  WiFi.begin(WIFI_SSID, WIFI_PASSWORD, 13, ap_mac, true); //reference: https://github.com/esp8266/Arduino/blob/master/doc/esp8266wifi/station-class.md#begin
}

void wifiSuccess() {
  Serial.print("---\nCONNECTED ! (");
  Serial.print(millis() - begints);
  Serial.println("ms)");
  //WiFi.printDiag(Serial); //enable for more details on WiFi connection status
  Serial.print("AP with BSSID : ");
  Serial.printf("BSSID: %s\n", WiFi.BSSIDstr().c_str());
}

void loop() {
  //displays connection status once connected
  bool bWiFi_connected = WiFi.status() == WL_CONNECTED;
  if (/*on connect*/ bWiFi_connected && !b_WiFi_connected_bkp) {
    wifiSuccess();
  }
  b_WiFi_connected_bkp = bWiFi_connected;
}
```

### Debug Messages

Using the correct BSSID:
```
Trying to connect to AP with BSSID: <MY AP BSSID>
---
CONNECTED ! (1322ms)
AP with BSSID : BSSID: <MY AP BSSID>
```

Using the wrong one:
```
Trying to connect to AP with BSSID: <BULLSHIT BSSID>
---
CONNECTED ! (15939ms)
AP with BSSID : BSSID: <MY AP BSSID>

```

schlaubstar, 2017-03-23:
Hi ... as you also seem to try to speed up connectivity just one thing I have encountered with two different ESP8266-01 (one the old, one the new -S Version).

My sketch looks like that and thereby gives me two options to connect (all parameters like MAC-Addresses, SSID, Password are just dummy ones in this code)

   ` const char* ssid="mySSID";
    const char* pwd="myWPA2Password";
   
    IPAddress ip(192,168,xxx,xx);
    IPAddress gw(192,168,xxx,1);
    IPAddress sn(255,255,255,0);

    //unsigned char ap_mac[18] = { 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF };  
   
    WiFi.persistent(false);    //avoids updating the same flash cell every reboot
    WiFi.config(ip, gw, sn);   
    WiFi.begin(ssid, pwd);
    //WiFi.begin(ssid, pwd, 13, ap_mac, true);`

What it also does is to read an Si7021 sensor, and submit the values to thingspeak.
When I use the code as it is above I get an average total runtime of < 500ms from wakeup to the next deepsleep cycle.

If I remove WiFi.begin(ssid, pwd) and replace it by the 'to be faster' version providing channel and ap_mac, the average total runtime raises to > 1.500ms.

I tried a lot of combinations and finally came to the conclusion, that it is the fastest way to not provide channel and ap_mac.

In my router, I have reserved the IP of the ESP8266-01 for it, to avoid this IP address is blocked or assigned to another device.

This does not really help you with the encountered "bug", but maybe the version is still faster than yours.
Give it a try.

schlaubstar, 2017-07-23:
I must add some aspect:
The solution you provided at the top (ignoring the strange fact that the BSSID seems to be half ignored) is faster than my proposal but it depends on the scenario:

I did some extensive tests with my ESP8266-12F:
Scenario1:
Loop:
  Turning WIFI on,
  Connecting
  Turning WiFi off,
  waiting

In this case the connection with BSSID and channel is approx done in 50ms (using average timings of 30 rounds, and a delay of 20ms waiting to get WL_CONNECTED status.
Connecting with SID and PWD only takes an average of ~ 500ms. (which makes your variant incredibly fast)

However: if I change the scenario to:

Loop:
  Turning WIFI on,
  Connecting
  Sending ESP to deepsleep

It does not make any difference if I try to connect with BSSID, channel and SSID or only SSID and PWD.
In both cases I get average connection times of about 160ms.

devyte, 2017-09-07:
A long time ago in a repo far far away, I witnessed a conversation about the BSSID. If memory serves (don't assume it does), the described behavior is meant to cover the case where you have several APs with the same SSID/password covering an area. Of course, in that case each ap has its own mac. In that scenario, if the ESP's configured ap drops out for whatever reason, but another one is still in range, you don't want the connection to fail, but rather fallback to the default behavior, and succeed with a different ap. In other words, let the ESP take a bit longer, but don't fail.

I find @schlaubstar findings most interesting, I hope it helps somebody to reduce wakeup/sleep cycles.

I don't think there is an issue here. Closing.

closed 2017-09-07 by devyte

JonathanCaes, 2018-08-28:
Hi
How do you know which channel to connect to if the AP is set to auto select the channel of communication? 
Thanks!

schlaubstar, 2018-08-29:
@JonathanCaes I don't know why you re open this thread for you question but it's simple: you don't. As your AP will auto-configure the channel depending on the signals of other networks (that will dynamically change and can not be affected by you) so you can not use a connection setup where you transmit the channel. However: As mentioned: connecting with station name (SSID) and password can be as fast as connecting with a full set of parameters