esp8266/Arduino #3029: How to instantiate WiFiClient or WiFiClientSecure at runtime?

Humancell, 2017-03-08:
Is there a way that at runtime I can instantiate a WiFiClient OR WiFiClientSecure based on the scheme specified in the URI?

I'm trying to do something like this, that obviously is wrong and won't work ... but I'm not quite clear on how I would do this?  I tried posting to the forums, but my post has been held in moderation for a while and I'm trying to get this working.

```
    // http or https?
    if(!strcmp(endpointScheme, "http")) {
      // Use WiFiClient class to create TCP connections
      WiFiClient httpClient;
      #ifdef DEBUG
      Serial.println(F("Using HTTP://"));
      #endif
    } else {
      // Use WiFiClientSecure class to create TCP connections
      WiFiClientSecure httpsClient;
      #ifdef DEBUG
      Serial.println(F("Using HTTPS://"));
      #endif
    }

    int connectResult;
    connectResult = client.connect(WiFi.gatewayIP(), atoi(endpointPort));

```

I've tried to also use a void pointer, but realize this is not correct either ... I'm trying to avoid having two completely redundant sets of code to handle this ... I keep thinking there is a way that I am compeltely missing?

I've tried this also ... no go.

```
    // http or https?
    WiFiClient httpClient;
    WiFiClientSecure httpsClient;
    ClientContext* client;
    
    if(!strcmp(endpointScheme, "http")) {
      // Use WiFiClient class to create TCP connections
      (WiFiClient) client = httpClient;
      #ifdef DEBUG
      Serial.println(F("Using HTTP://"));
      #endif
    } else {
      // Use WiFiClientSecure class to create TCP connections
      (WiFiClientSecure) client = httpsClient;
      #ifdef DEBUG
      Serial.println(F("Using HTTPS://"));
      #endif
    }

    int connectResult;
    connectResult = client.connect(WiFi.gatewayIP(), atoi(endpointPort));

```

gpepe, 2017-03-09:
something like that:
WiFiClient* client = (ssl) ? new WiFiClientSecure() : new WiFiClient();
............
delete client;

closed 2017-05-15 by igrr

elpiloto100, 2017-06-04:
Did this work for you? I tried this for a connection that requires WiFiClientSecure, and it didn't work. In fact, if I just try
`WiFiClient client = WiFiClientSecure();`
It fails as if I am using just WiFiClient.

devyte, 2018-08-29:
One of the downsides of polymorphism is that you don't have direct access to the methods specific to the derived classes, so no, you can't just call verify() because it's specific to the secure one.
If you still have the ssl flag available, use that to condition your call to verify. If not, a simple workaround is to use something like this:
```
WiFiClientSecure *secure = dynamic_cast<WiFiClientSecure>(client);
if(secure)
  verify();
```that

Please be aware that the default WiFiClientSecure that you're using is built over the axtls lib, which is no longer maintained. It it is therefore planned for deprecation and then retirement. There is an alternative in our core called bearssl::WiFiClientSecure which is currently recommended for use. So far it has proven to be more reliable than the default axtls one, but it is still rather new so it's considered experimental. The api is almost-but-not-quite the same,  see available examples for usage.

Humancell, 2018-08-29:
Thank you for the response!  I was just about to post the solution that *appears* to maybe be working?

```
    if(!useHttp) {
      // create new pointer cast as secure ...
      WiFiClientSecure* httpsClient = (WiFiClientSecure*)httpClient;
      if (httpsClient->verify(fingerprint, endpointHost.c_str())) {
        Serial.println(F("certificate matches"));
      } else {
        Serial.println(F("certificate doesn't match"));
      }
    }

```
I realized that I would have to do some sort of casting to allow a reference to the secure methods.  I was having trouble determining the syntax.  @devyte could you comment on your method vs. what I got to compile?    I'm guessing that using `dynamic_cast<WiFiClientSecure>(client)` is the "more proper" way to accomplish this?

devyte, 2018-08-29:
The dynamic_cast call will fail if the argument can't be cast to the intended type and then returns nullptr. It is the "safer" cpp way.