esp8266/Arduino #3030: SPIFFS.begin() crashes (stops) ESP without errors and only a line of garbage in serial monitor

pjunni, 2017-03-09:
### Basic Infos

#### Hardware
Hardware:			ESP-07
Core version: I don't know

### Description

All I get in the serial monitor is this:

> rrlaz|ela|Œlìb|Ž‚ì’r’bŒbŒònnžlnnœâìbpŒŽlrlrl‚rò’nàbânbp

When I should get (if ESP not damaged):

> OKOK?

Or if damaged I should get an error message or equivalent that SPIFFS begin fails. If I remove "SPIFFS.begin();" The sketch runs normally.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	80Mhz
Flash Mode:		dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

````
#include "FS.h"

void setup() {
  Serial.begin(115200);
  Serial.println("OKOK?");
  SPIFFS.begin();
}

void loop() {
}
````

pjunni, 2017-03-09:
Sorry to spam here. The solution was to upload with correct flash size settings in the IDE. To enable error messages and prevent this kind of issue here is one solution to the code above:

````
#include "FS.h"

String realSize = String(ESP.getFlashChipRealSize());
String ideSize = String(ESP.getFlashChipSize());
bool flashCorrectlyConfigured = realSize.equals(ideSize);

void setup() {
  Serial.begin(115200);
  if(flashCorrectlyConfigured) SPIFFS.begin();
  else Serial.println("flash incorrectly configured, SPIFFS cannot start, IDE size: " + ideSize + ", real size: " + realSize);
}

void loop() {
}
````

closed 2017-03-09 by pjunni

suculent, 2017-08-11:
I have exactly the same issue even using this safety code. SPIFFS.format() works but SPIFFS.begin() crashes with nothing interesting in stacktrace.

Similar to issue https://github.com/esp8266/Arduino/issues/2655

sovcik, 2019-02-27:
I recently (today) had the same issue with ESP07. 
Tried with two chips to make sure it is not MC related issue.

And the result was: **power supply**.

Issue disappeared after adding 1mF capacitor to the circuit ...

diegogrosmann, 2019-04-25:
Tive o mesmo problema. Estava escolhendo o FlashSize "1M (no SPIFFS)", alterei para "1M (192 SPIFFS)" e funcionou.

suculent, 2019-04-26:
This makes sense. Can’t use SPIFFS without having one.

Isso faz sentido. Não é possível usar o SPIFFS sem ter um.

Sorry for my Google-Portuguese but I just wanted others to know what is this about.

> On 26 Apr 2019, at 00:05, diegogrosmann <notifications@github.com> wrote:
> 
> Tive o mesmo problema. Estava escolhendo o FlashSize "1M (no SPIFFS)", alterei para "1M (192 SPIFFS)" e funcionou.
> 
>

dg321, 2019-04-26:
O único cuidado que devemos ter é com o tamanho do SPIFFS. Se estiver usando OTA o tamanho total do bin (sketch + SPIFFS) tem que ser menor que 50% da FLASH.

The only care we should take is with the size of SPIFFS. If using OTA, the total bin size (sketch + SPIFFS) must be less than 50% of FLASH.

suculent, 2019-04-26:
Actually, this issue is caused by the fact, that default board setting for certain (most) devices is No SPIFFS. 

I’m dealing usually with 32mb/4MByte devices where the sketch takes less than 1MB which is OK for OTA when the SPIFFS is minimal (2MB/Mo = 16Mb plus 1MB sketch and 1MB sketch temp for ota switch. 

On 

M. 

Odesláno z iPhonu