esp8266/Arduino #3035: Can not download esp8266 board manager url (again) (again)

rhtenhove, 2017-03-11:
#1689 #2644 #2664

The webserver at arduino.esp8266.com is not working, and that has happened quite a few times before looking through the issue tracker.

Can you look into it?

Perhaps you can store these binaries on GitHub, and as such have much less reliance on whatever service or device you're using now.

gbr1, 2017-03-11:
we need it for a competition, please give us also a github, drive, dropbox, mega version of binaries

benjenq, 2017-03-11:
Same issue!
Try this:
https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json

schnti, 2017-03-11:
Error downloading http://arduino.esp8266.com/osx-xtensa-lx106-elf-gb404fb9-2.tar.gz

gfvalvo, 2017-03-11:
Thanks @benjenq !!!
Just add this to the end of "Additional Board Manager URLs"?
Will I need to remove it once arduino.esp8266.com server is back up?

rhtenhove, 2017-03-11:
Since that .json refers to binaries that are only available from arduino.esp8266.com, in this case the xtensa package, adding that json link instead is fruitless.
Same goes for the git installation method, since the get.py also refers to arduino.esp8266.com.

gfvalvo, 2017-03-11:
The link @benjenq provided is working for me. I can now compile for Huzzah ESP8266 on Arduino 1.6.12.

leemoore1966, 2017-03-11:
Can anyone help here, seems all options fail in some way or another, I am now getting
Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz

caccac, 2017-03-11:
same problem here. After adding additional board url as https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json , esp8266 package was seen on Board Manager.

Bur after try to install it's giving "Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz" error.

yorik, 2017-03-11:
The site arduino.esp8266.com (104.131.82.128) is down. Any mirrors?

ABIV, 2017-03-11:
Mirrror would be nice, as it's been noted this isn't the first time it's down.

RothBoy, 2017-03-11:
@igrr any chance you can take care of this?

BoSsOfDMaX, 2017-03-11:
Same here,

same problem here. After adding additional board url as https://github.com/esp8266/Arduino/releases/download/2.3.0/package_esp8266com_index.json , esp8266 package was seen on Board Manager.

But after try to install, it's giving "Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz" error.

Fbo06, 2017-03-11:
it's crazy that no mirror take place. it's not the first time ! i really think now to use another board because i can't be an hostage

BoSsOfDMaX, 2017-03-11:
Githun link is working but when try to install it gives _"Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz"_ error.

I tryed to install on Arduino version 1.6.13 but same problem

BoSsOfDMaX, 2017-03-11:
Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
java.lang.RuntimeException: java.lang.Exception: Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more
Caused by: java.net.ConnectException: Connection refused: connect
	at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:85)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 4 more

RothBoy, 2017-03-11:
up and running again :-)

caccac, 2017-03-11:
yes. tested. 

now downloaded and installed succesfully.

BoSsOfDMaX, 2017-03-11:
Works perfectly, Thank you :)

AmritSd, 2017-03-14:
Still not working for me

AmritSd, 2017-03-14:
Somebody help pls.

kathar1223, 2017-03-14:
Even it's not working for me too..`Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
java.lang.RuntimeException: java.lang.Exception: Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Error downloading http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:69)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:110)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more`

kathar1223, 2017-03-14:
@igrr , @RothBoy guys please look into this and bring up it soon

Thanks,
kamran

igrr, 2017-03-14:
Seems to work here, both using curl and web browser... can you try opening the url with curl and check what kind of error it reports (i.e. connection refused, connection timeout, some http error)?

kathar1223, 2017-03-14:
@igrr , able to download using web.. and even i see in arduino it's downloading now as it is of more than 100mb.. will let you know if i see any issues.. Thanks @igrr

AmritSd, 2017-03-14:
It's working again! Thanks everybody for the help.

noahziheng, 2017-03-21:
Maybe the website is down again...
2017.3.21 05:30UTC

ancientmidi, 2017-03-21:
Same issue.

per1234, 2017-03-21:
At the request of @igrr:
Tried right after @noahziheng's comment:
```
C:\>curl http://arduino.esp8266.com/stable/package_esp8266com_index.json
curl: (56) Recv failure: Connection was reset
```
Tried just before this comment:
```
C:\>curl http://arduino.esp8266.com/stable/package_esp8266com_index.js
curl: (7) Failed to connect to arduino.esp8266.com port 80: Timed out
```

rexopl, 2017-03-21:
Looks like http://arduino.esp8266.com/ is down again. Any chance to bring it up?
For json we can use this link: https://github.com/esp8266/Arduino/releases/download/2.1.0/package_esp8266com_index.json
But still xtensa is missing.

igrr, 2017-03-21:
It looks like the CPU ran away to do some interesting things instead of serving all those json files.
Should be up now.

tomyerl, 2017-03-30:
Hey everyone, seems like there is a new issue in town:

Gateway Anti-Virus Alert

This request is blocked by the Firewall Gateway Anti-Virus Service. Name: Pirminay.A_19 (Trojan)

Help?

forumengine, 2017-04-11:
Возникла ошибка при загрузке http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Возникла ошибка при загрузке http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:151)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: Возникла ошибка при загрузке http://downloads.arduino.cc/packages/package_index.json
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:131)
	at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:334)
	at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:309)
	at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:287)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$0(ContributionManagerUI.java:147)
	... 1 more
Caused by: java.net.SocketTimeoutException: connect timed out
	at java.net.DualStackPlainSocketImpl.waitForConnect(Native Method)
	at java.net.DualStackPlainSocketImpl.socketConnect(DualStackPlainSocketImpl.java:85)
	at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
	at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
	at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
	at java.net.PlainSocketImpl.connect(PlainSocketImpl.java:172)
	at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
	at java.net.Socket.connect(Socket.java:589)
	at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:432)
	at sun.net.www.http.HttpClient.openServer(HttpClient.java:527)
	at sun.net.www.http.HttpClient.<init>(HttpClient.java:211)
	at sun.net.www.http.HttpClient.New(HttpClient.java:308)
	at sun.net.www.http.HttpClient.New(HttpClient.java:326)
	at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1202)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1181)
	at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1032)
	at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:966)
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:170)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:128)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:129)
	... 5 more
pls halp with add esp8266

forumengine, 2017-04-11:
not possible to add esp8266 board, please somebody halp to me?

Eliaxie, 2017-05-04:
Same for me, cannot download http://arduino.esp8266.com/win32-xtensa-lx106-elf-gb404fb9-2.tar.gz

igrr, 2017-05-04:
Moving to a new issue: https://github.com/esp8266/Arduino/issues/3198.

closed 2017-05-04 by igrr