esp8266/Arduino #3036: espcomm_sync failed

EBlnr, 2017-03-11:
Hello,

I'm just starting using Arduino and i would like to realise this : http://iot-playground.com/blog/2-uncategorised/41-esp8266-ds18b20-temperature-sensor-arduino-ide

I used their code which is OK when verified, but i have problem when i try to upload it. Here are my connections on the pins of the ESP : 

TXD - RXD of CP2102
RXD - TXD of CP2102
VCC - 3.3V (external power supply)
GND - GND (external power supply) 
GPI00 - GND
GPI02 - GND 
(GPI015 - GND) -> not sure for that one but i tried it because some people gave that solution on this forum
CH_PD - 3.3V

I saw that many other people have that kind of mistakes and solutions have often been given, but the fact is that i tried a lot of them and it didn't work, including : 
- Changing 3.3V for a external power supply instead of USB
- Inverting TXD and RXD
-GPI02 on 3.3V instead of GND

I spend hours but without success so please let me know if you have any idea to solve it.

Thank you.

#### Hardware
Hardware:			ESP8266-ESP08
Core Version:      	?

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		?
Reset Method:		ck 

### Debug Messages

```
Arduino : 1.6.12 (Windows 10), Carte : "Generic ESP8266 Module, 80 MHz, 40MHz, DIO, 115200, 512K (64K SPIFFS), ck, Disabled, None"

warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

More details :

	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
...

```

vaishnavi-28, 2017-06-24:
Same problem occurred to me as also. Please help me if you have found out the solution for it!!

devyte, 2017-09-07:
The message error: espcomm_open failed is indicative that the COM port couldn't be opened. That is usually the case when the drivers are missing, wrongly installed, or some other issue with the serial port.
This is not the right place to requesting assistance with IDE usage. This is an issue tracker, meant for tracking issues in the core code. Please refer to esp8266.com or stackoverflow.
Closing.

closed 2017-09-07 by devyte