esp8266/Arduino #3038: axTLS is shipped as a binary blob without attribution

micolous, 2017-03-13:
https://github.com/esp8266/Arduino/commit/e8b8a606e4fa8df510b458507b15ee332317f410

It looks like you're shipping axTLS as a binary blob. That should really be built from source, such that anyone can update it.  Consider using `git submodule` in order to pull in the source from upstream.

The code is also [BSD licensed](http://axtls.sourceforge.net/README/index.html), and the copyright doesn't appear to be mentioned anywhere in the product documentation apart from the `ssl.h` header file.

igrr, 2017-03-13:
Thanks for pointing this out, I'll update the readme file to add information about axTLS use and its license.
The issue with building axTLS from source is that it prevents the repository from being usable via "git clone into Arduino hardware folder" method. One would also need `make` to be installed to build axTLS, and unfortunately `make` is not available on all platforms by default.

micolous, 2017-03-13:
Maybe there should be a binary release from this repository instead (IE:
using tags) for those that don't have a build environment available?

That way those without make can grab a ZIP file, and not need to worry
about even installing git.

igrr, 2017-03-13:
There are tagged releases (see releases page of the repository), which can be installed from the boards manager in the IDE.

closed 2017-03-13 by igrr