esp8266/Arduino #3039: NodeMCU ESP8266 cannot start

tranquang7979, 2017-03-13:
I'm building a small project that user can control 2 leds through the wifi. After I uploaded program to NodeMCU ESP8266. I got these messages

[image]

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			NodeMCU ESP8266 MOD

### Description

[image]

### Settings in IDE

[image]

### Sketch

`#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
 
MDNSResponder mdns;
 
ESP8266WebServer server(80);
String webPage;
const char* ssid     = "Tran Gia";      //Thay tên wifi ở đây
const char* password = "0937468606";  //Thay tên password ở đây
int led1 = 6;//12
int led2 = 7;//13

void setup() {
 
  pinMode(led1, OUTPUT);  //led 12
  pinMode(led2, OUTPUT);  //led 13
 
  webPage += "<h1>ESP8266 Web Server</h1><p>Socket #1 <a href=\"socket1On\"><button>ON</button></a>&nbsp;<a href=\"socket1Off\"><button>OFF</button></a></p>";
  webPage += "<p>Socket #2 <a href=\"socket2On\"><button>ON</button></a>&nbsp;<a href=\"socket2Off\"><button>OFF</button></a></p>";
  
  Serial.begin(115200);
  delay(100);
 
 
  Serial.println();
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
 
  Serial.println("");
  Serial.println("WiFi connected");  
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  
  if (mdns.begin("esp8266", WiFi.localIP())) 
    Serial.println("MDNS responder started");
 
  server.on("/", [](){
    server.send(200, "text/html", webPage);
  });
  server.on("/socket1On", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(led1, HIGH);
    delay(1000);
  });
  server.on("/socket1Off", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(led1, LOW);
    delay(1000); 
  });
  server.on("/socket2On", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(led2, HIGH);
    delay(1000);
  });
  server.on("/socket2Off", [](){
    server.send(200, "text/html", webPage);
    digitalWrite(led2, LOW);
    delay(1000); 
  });
  server.begin();
  Serial.println("HTTP server started");
}
 
void loop() {
  server.handleClient();
}`

tomamarsanaa, 2017-03-19:
Hi, I had same problem and I have spent many weeks to try to solve this issue.
At last, I found out the reason. Try to set 16th pin mode in the 'setup' function as below.
Doing this, mine is resolved.

void setup() {
  pinMode(16, OUTPUT); // set mode for build in LED which is 16th pin
  ...
}

devyte, 2017-09-07:
GPIOs 6 & 7 are used internally, and shouldn't be used for other purposes. Don't confuse GPIO6 & 7 with D6 & 7.
Please research more about the nodemcu and pinouts. 
Closing as user error.

closed 2017-09-07 by devyte