esp8266/Arduino #3040: HTTPS client.connect issue

coyt, 2017-03-13:
Alright guys, I'm trying to pull some Json data from a few different websites to make a physical bitcoin / ethereum ticker. 

I can successfully connect and retrieve data from every HTTPS site I've tried except one:

[https://api.nanopool.org/v1/eth/prices](https://api.nanopool.org/v1/eth/prices)

For the above, the example WiFiClientSecure code fails at "client.connect(host, httpsPort)" 

This HTTPS site works fine as well as all others i've tried:

[https://coinmarketcap-nexuist.rhcloud.com/api/eth](https://coinmarketcap-nexuist.rhcloud.com/api/eth)

I'm using Arduino 1.8.1, esp8266 version 2.3.0 from Arduino Boards Manager, and a nodeMCU 1.0 board. I've tried making this work in Atom and Clion with platform.io as well with the same problem. 

I've posted my code below as well as the debug messages I'm getting. Could this be the CA cert size issue I've been reading about on similar issues? Thanks for the help!

I normally use defualts for NodeMCU 1.0 in the IDE but I switched to generic for the debug. Same problem was present in both. 
Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Debug Level: Core + WiFi
Reset Method:		nodemcu

### Sketch
/*
 *  HTTP over TLS (HTTPS) example sketch
 *
 *  This example demonstrates how to use
 *  WiFiClientSecure class to access HTTPS API.
 *  We fetch and display the status of
 *  esp8266/Arduino project continuous integration
 *  build.
 *
 *  Created by Ivan Grokhotkov, 2015.
 *  This example is in public domain.
 */

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>

const char* ssid = "HellSpot Slow";
const char* password = "ILikeWiFi";

//const char* host = "coinmarketcap-nexuist.rhcloud.com"; //works fine
const char* host = "api.nanopool.org"; //does NOT work

const int httpsPort = 443;

void setup() {
    Serial.begin(115200);
    Serial.setDebugOutput(true);
    Serial.println();
    Serial.print("connecting to ");
    Serial.println(ssid);
    WiFi.begin(ssid, password);
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }
    Serial.println("");
    Serial.println("WiFi connected");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());

    // Use WiFiClientSecure class to create TLS connection
    WiFiClientSecure client;
    Serial.print("connecting to ");
    Serial.println(host);
    if (!client.connect(host, httpsPort)) {
        Serial.println("connection failed");
        return;
    }

    //String url = "/api/eth"; //works fine
    String url = "/v1/eth/prices"; //does NOT work
    Serial.print("requesting URL: ");
    Serial.println(url);

    client.print(String("GET ") + url + " HTTP/1.1\r\n" +
                 "Host: " + host + "\r\n" +
                 "User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/56.0.2924.87 Safari/537.36\r\n" +
                 "Connection: close\r\n\r\n");

    Serial.println("request sent");
    while (client.connected()) {
        String line = client.readStringUntil('\n');
        if (line == "\r") {
            Serial.println("headers received");
            break;
        }
    }
    String line = client.readStringUntil('\n');
    if (line.startsWith("{\"state\":\"success\"")) {
        Serial.println("esp8266/Arduino CI successfull!");
    } else {
        Serial.println("esp8266/Arduino CI has failed");
    }
    Serial.println("reply was:");
    Serial.println("==========");
    Serial.println(line);
    Serial.println("==========");
    Serial.println("closing connection");
}

void loop() {
    Serial.println("testing");
    delay(1000);
}

##DEBUG MESSAGES (SERIAL OUTPUT):
connecting to HellSpot Slow
sta config unchangedscandone
f 0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 7
cnt 

connected with HellSpot Slow, channel 10
dhcp client start...
wifi evt: 0
.ip:192.168.29.143,mask:255.255.255.0,gw:192.168.29.1
wifi evt: 3
.
WiFi connected
IP address: 
192.168.29.143
connecting to api.nanopool.org
[hostByName] request IP for: api.nanopool.org
[hostByName] Host: api.nanopool.org IP: 104.18.43.243
:ref 1
please start sntp first !
:wr
:sent 79
:ww
:rn 7
:rcl
:abort
:rd 5, 7, 0
:rdi 7, 5
:rd 2, 7, 5
:rdi 2, 2
:c0 2, 7
connection failed
:ur 1
:del
testing
testing
testing
testing
testing

brunohorta82, 2017-07-27:
I have the same problem :(

cirorizzo, 2017-10-09:
Either I do

devyte, 2018-05-29:
BearSSL is merged in #4273 , with alternate BearSSL::WiFi* classes. Although axtls-based classes are still available and even the default, they are planned for deprecation and then retirement, hence won't be fixed. Any issues with BearSSL-based classes should be reported in new issues.
Closing.

closed 2018-05-29 by devyte