esp8266/Arduino #3042: Yield and Delay

hreintke, 2017-03-13:
Hi,

My application is mostly async and uses tickers and other callbacks to facilitate that.

Question on yield()

In these callbacks, the yielding is not possible. 
When one of the used libraries/framework functions is calling the yield() function there is a "panic" stop.

An example of this is the hexdump function within debug.cpp

Is there a specific reason for the "panic exit" instead of "just return" when "cont_can_yield(&g_cont)" returns false ? 

Question on delay()

I know there should be no blocking functions in but some libs/functions use a delay(1) or delay(2) to send a pulse.
In the delay function the esp_yield() is used but when cont_can_yield(&g_cont) returns false (in callback) there is no delay at all.
The application then crashes.

An update to the code below would prevent that, any comment to that ? 

```
void delay(unsigned long ms) {
    if (cont_can_yield(&g_cont))
    {
    	if(ms) {
          os_timer_setfn(&delay_timer, (os_timer_func_t*) &delay_end, 0);
          os_timer_arm(&delay_timer, ms, ONCE);
		} else {
           esp_schedule();
		}
		esp_yield();
		if(ms) {
			os_timer_disarm(&delay_timer);
		}		
	}
    else
    {
    	delayMicroseconds(ms*1000);
    }
}
```

devyte, 2017-09-03:
I am not an expert on this matter, but I have looked a bit into the behavior, and I have also followed some conversations related to the topic.
Therefore, I am goin to hazard some educated answers to your questions.

> is there a specific reason for the "panic exit" instead of "just return"

If a function uses yield, it is assumed that it requires it to function properly. Given the limitations of yield, it MUST therefore run inside the loop. The reason for the panic is to forcibly detect and disallow calls to yield from outside the loop.

Yield works by giving up the context inside the loop and pass it to the code outside of it, which then runs a function that services the wifi stack. Once that service is done, it returns to the stored context. If you are running async callbacks, I believe (I am not 100%sure of this) you are running from within the context of the espressif sdk, which means you are already running from within the "wifi service routine". Therefore, yield can't work, because it would yield its context to itself, and then have to return to itself.

> An update to the code below would prevent that

Your change makes use of delayMicroseconds. That essentially hogs the cpu doing nothing, which in the ESP is a Bad Idea, because during that time the wifi stack goes unserviced, and could blow up the internal buffers. In the async world, there is a guideline for maximum time that user code can be away from servicing the wifi, which I believe is about 15ms in total (if memory serves, I would have to read some docs to make sure).
In short, the code would have a radically different behavior when called from the loop vs. when called from an async callback, with different potential crash reasons and a whole lot of instability. Just to make you aware, even different environments, i.e.: lots of nearby wifi points, or many devices on the same wifi network, have an impact, because such factors stress the wifi servicing on the ESP. Adding this on top makes that a lot worse.

Personally, I am also using async callbacks, and I have already had two cases where I've had to deal with libs that had yields and delays in them. One case is reading a DHT, where I ended up reading the sensor from the loop, and the async interface would communicate and control through a global context. The other was a Stream object in the core, and in that case I just removed the calls to yield from Stream.cpp. That's a bold move, it works for my own app, but I don't think the code change should be merged, because I doubt it will work for big files or data chunks.
In case you're wondering, yes, I did that because I had to manipulate files from within AsyncWebServer request callbacks, that had nothing to do with serving files to the http client. The files I manipulated were tiny, always under 2KB, and I read/wrote them in one pass to/from memory.

My suggestion to you: implement the changes in your own local git of the libs, and keep them there. If necessary, pull them out of your local repo to update the git version, then add them back on top. Test them, and maybe document them somewhere public. You can even report back here, even after the issue is closed. Whatever else, be VERY aware that they can cause random crashes.

Having said all that, I would say the current state of the code is a design choice, not an issue, in which case please close this. If you disagree, we can wait for somebody else to chime in, orwe can try to get somebody more knowledgable to give an opinion (doubtful with 1000 issues open atm).

devyte, 2017-09-07:
No further comments in 5 days. Closing.

closed 2017-09-07 by devyte