esp8266/Arduino #3044: ESP8266-WPS error

vivky, 2017-03-13:
This is the error I got while trying to compile my ESP8266MOD.......Can anyone suggest me how to rectify these error......
/Users/[MYCOMPUTER]/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp: In function 'void wifi_wps_status_cb(wps_cb_status)':
/Users/[MYCOMPUTER]/Documents/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:593:14: error: 'WPS_CB_ST_UNK' was not declared in this scope
case WPS_CB_ST_UNK:
^

devyte, 2017-09-07:
@vivky no basic info, no sketch, no build system or development details, no usable info to reproduce.
I can vouch for there currently being no compile issues with that file.
Closing due to lack of info and likely user error.

closed 2017-09-07 by devyte

riker65, 2017-11-14:
Hi have the same problem

compile error
/Arduino/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:599:14: error: 'WPS_CB_ST_UNK' was not declared in this scope
         case WPS_CB_ST_UNK:

sketch:

//Connect ESP8266 to Wi-Fi Router
//Upload these code to your Arduino WeMos D1 ESP8266 W-Fi board.
//http://aconcaguasci.blogspot.de/2017/01/connect-arduino-wemos-d1-esp8266-to.html

#include <ESP8266WiFi.h>

//SSID of your network
char ssid[] = "myssid"; //SSID of your Wi-Fi router
char pass[] = "mypw"; //Password of your Wi-Fi router

void setup()
{
  Serial.begin(115200);
  delay(10);

  // Connect to Wi-Fi network
  Serial.println();
  Serial.println();
  Serial.print("Connecting to...");
  Serial.println(ssid);

  WiFi.begin(ssid, pass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("Wi-Fi connected successfully");
}

void loop () {}