esp8266/Arduino #3047: Strange output on serial port when reset ESP

tomaszmacnar, 2017-03-14:
### Basic Infos

#### Hardware
Hardware:			ESP-07
Core Version:      	2.3.0

### Description

When upload any program (for example Blink) I have strange output (like in screen) on serial port. I need to have clean console. How to fix it ?
[image]

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		no_ide_how_it_check
Reset Method:		ck

lrmoreno007, 2017-03-15:
It is normal in ESP8266 and is independent of the framework you use. Espressif says it can not be disabled.

But you can try transmitting with Serial1 as TX and receiving Serial as RX.
https://github.com/esp8266/Arduino/blob/master/doc/reference.md#serial

tomaszmacnar, 2017-03-15:
Thanks for answear. I will check this solutions soon :)

davisonja, 2017-03-15:
For completeness:
It's a start up message from the espressif bootloader and is at 74880 baud.
Or the baud rate set for a soft reset (as opposed to a power cycle)

or mute
the thread

lrmoreno007, 2017-03-15:
No, it's a boot log from the ROM. For that reason Espressif can't deactivate it, at least that's what they say.

davisonja, 2017-03-15:
Yes, the ROM bootloader. Which, among other things, outputs log information on the reset cause and the boot mode.

> https://github.com/espressif/esptool/wiki/ESP8266-Boot-ROM-Log

covers it.

devyte, 2017-09-07:
Those messages can't be avoided on the main serial port. If you need a clean startup, please follow the suggestions above.
Closing.

closed 2017-09-07 by devyte

SergiuToporjinschi, 2019-07-11:
That is a serious problem. I spent a week debugging a circuit of a DHT and ESP8266-01 and because I was using GPIO2 (which is also a serial TXD1) and the esp is throwing this boot message on all Serials it seems. So, on ESP8266-01 only one pin available, and that is GPIO-0