esp8266/Arduino #3048: I2C trasmission work with Arduino but failed with ESP8266

muke1970, 2017-03-15:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description
I must set some registers of a chip using I2C. It work good with Arduino but fail with ESP12-F 

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch with Arduino
#include <Wire.h>
void setup() {
  Serial.begin(115200);
  Wire.begin();
  Wire.beginTransmission(0x20);         
  Wire.write(0x2D);              
  Wire.write(0x43);      
  Wire.write(0x4D);
  Wire.write(0x53);  
  Wire.write(0x03);              
  Wire.endTransmission();
  delay(10);
}

void loop() {
  Serial.println("Set ok");
  delay(1000);
}

### Sketch with ESP8266
#include <Wire.h>
void setup() {
  Serial.begin(115200);
  Wire.begin(5,4); 
  Wire.setClockStretchLimit(1500);
  Wire.beginTransmission(0x20);         
  Wire.write(0x2D);              
  Wire.write(0x43);      
  Wire.write(0x4D);
  Wire.write(0x53);  
  Wire.write(0x03);              
  Wire.endTransmission();
  delay(10);
}

void loop() {
  Serial.println("Set ok");
  delay(1000);
}

### Debug Messages

```
I don't know if it's a problem with stretching clock but on ESP8266 doesn't work. The chip I use on I2C is the Cypress CY8C9520A. Other type of trasmission to that chip work like a charm on Arduino and ESP8266. I don't know if a bug but I'm not able to solve the problem.
Can you help me?
```

Wafalafel, 2017-03-22:
You made a mistake in calling Wire.begin(5,4) 
The correct call should be Wire.begin(4,5)

see link below
https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#i2c-wire-library
"......
Wire library currently supports master mode up to approximately 450KHz. Before using I2C, pins for SDA and SCL need to be set by calling Wire.begin(int sda, int scl), i.e. Wire.begin(0, 2) on ESP-01, else they default to pins 4(SDA) and 5(SCL).
....."

muke1970, 2017-03-23:
Thanks for your reply.
I use Wire.begin(5,4) because in my PCB design i make a mistake and I reversed SDA with SCL (other function working good on I2C).
On ESP12, using SDA(pin 4) and SCL(pin 5), you can use Wire.begin() instead Wire.begin(4,5) because you are using defaults.
I suppose my problem is the clock stretching but I'm not able to solve it.

Wafalafel, 2017-03-23:
then you should go backto the default Wire.begin() call and try to solve your problem from there. I am sorry I cannot help you with the clock stretching issue.

devyte, 2017-09-07:
@muke1970 is this issue still relevant?

muke1970, 2017-09-12:
I didn't pursue the case because i use that function to initialize the chip only first time and i have some chips initialized ;-)

sneakyimp, 2020-06-20:
> @muke1970 is this issue still relevant?

Yes. I have two NodeMCU ESP8266 dev boards and I cannot get them to communicate using Wire. Others are having this issue also: https://github.com/esp8266/Arduino/issues/3046

Tech-TX, 2020-11-10:
@sneakyimp would you tell me which chip(s)  you can't talk to? If it's something I have or can get cheaply,  I can find the issue.  The change to clock stretch last year should have minimized most of the comms problems.

5chufti, 2020-11-15:
in the opening post he mentions Cypress CY8C9520A (a port expander)

earlephilhower, 2021-02-14:
Closing as the original issue (clock stretching support on master) is now in the core.

closed 2021-02-14 by earlephilhower