esp8266/Arduino #3049: POST requests are returning extra characters around JSON response

dbudwin, 2017-03-15:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3

### Description

When making REST requests to my API, the returned JSON response consistently has the same extra characters before and after the response ("10" before body and "0" after the body, without the quotes).  Testing my API using both Advance REST Client and Hurl.it demonstrate that my API works correctly and that the source of the extra characters isn't from the API and that it's not an issue with the headers/parameters I am sending.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4M (3M SPIFFS)
CPU Frequency:		80Mhz
Flash Mode:			?
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		?

### Sketch

```
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <RCSwitch.h>
#include <RestClient.h>

const char* ssid = "...";
const char* password = "...";
const char* mqttChannel = "...";
const char* mqttServer = "...";
const char* mqttUser = "...";
const char* mqttPassword = "...";
const int mqttPort = 12345;

RCSwitch rcSwitch = RCSwitch();

WiFiClient espClient;
PubSubClient client(espClient);

RestClient restClient = RestClient("somewebsite.com", 443, "My fingerprint");

void setupWifi() {
    delay(10);

    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(100);
    }
}

void callback(char* topic, byte* payload, unsigned int length) {
    String message = "";
    
    for (int i = 0; i < length; i++) {
        message += (char)payload[i];
    }

    rcSwitch.enableTransmit(0);
    rcSwitch.setPulseLength(184);

    restClient.setContentType("application/x-www-form-urlencoded");

    String response = "";

    int statusCode = restClient.post("/myPath", "my parameters", &response);

    StaticJsonBuffer<200> jsonBuffer;

    //THIS IS WHERE THE PROBLEM IS "response" WILL HAVE EXTRA CHARACTERS AROUND IT
    //SEE DEBUG SECTION BELOW TO SEE WHAT THIS OUTPUTS

    Serial.println("Response: >" + response + "<");

    JsonObject& root = jsonBuffer.parseObject(response);

    if (!root.success()) {
        Serial.println("parseObject() failed");
        return;
    }

    const char* sensor = root["code"];

    rcSwitch.send(12345, 24);
}

void reconnect() {
    while (!client.connected()) {
        if (client.connect("", mqttUser, mqttPassword)) {
            Serial.println("MQTT connected!");
            client.subscribe(mqttChannel);
        } else {
            Serial.print("Failed to connect to MQTT broker, rc=");
            Serial.print(client.state());
            delay(5000);
        }
    }
}

void setup() {
    Serial.begin(115200);
    setupWifi();
    client.setServer(mqttServer, mqttPort);
    client.setCallback(callback);
    rcSwitch.enableReceive(0); //Pin 3 on NodeMCU
}

void loop() {
    if (!client.connected()) {
        reconnect();
    }
    
    client.loop();
}
```

### Debug Messages

The output of the `response` println statement from my sketch above.  Note the `>` and `<` are mine and I expect them to be there:

```
Response: >10
{"code":5592835}
0

<
```

However, what I want is:

```
Response: >{"code":5592835}

<
```

I am using `WifiClientSecure` through an [ESP-8266 REST Client](https://github.com/DaKaZ/esp8266-restclient).

davisonja, 2017-03-15:
Does the '10' change if your JSON gets bigger? (is the '10' related to the
length of the response)
Have you turned on the REST Client debug?

dbudwin, 2017-03-15:
Good idea!  Yes the `10` does change.  When I made my function return `abcdefghijklmnopqrstuvwxyz0123456789` I get the following response:

```
Response: >2f
{"code":"abcdefghijklmnopqrstuvwxyz0123456789"}
0

<
```

Yes, I am running the REST Client in debug.  I'm fairly confident it's working okay since when in debug mode, the REST Client wraps `.,` around each call to `WifiClientSecure`'s `read()` function.  See line 222: https://github.com/DaKaZ/esp8266-restclient/blob/master/RestClient.cpp

davisonja, 2017-03-15:
It looked like the first number is the (clearly hex) number of bytes in the
response, the 0 meaning the end of the stream. It feels like HTTP chunked
encoding, or such.
Here it is: https://en.wikipedia.org/wiki/Chunked_transfer_encoding
Very useful for persistent HTTP connections (which require HTTP 1.1, so a
simple way to turn it off might be to specify HTTP 1.0).

Arguably it could be handled by the ESP REST library?

devyte, 2017-09-07:
From what I can tell, your use of the RestClient.post() call looks correct. The message that immediately follows it should be as you say: without the trailing chars.
However, the RestClient seems to be an external lib, so you should post the issue there, assuming it is still relevant and hasn't been fixed in the interim.
Closing, as this is not an issue in the core.

closed 2017-09-07 by devyte

HankLloydRight, 2022-02-06:
> It looked like the first number is the (clearly hex) number of bytes in the response, the 0 meaning the end of the stream. It feels like HTTP chunked encoding, or such. Here it is: [...]

I just wanted to drop in here and thank you for this.  I've been knocking my head against the wall for weeks trying to figure out why my ESP32 code using  WiFiClientSecure to pull content from my website, I was getting a 4-digit hex code _between_ the headers and the content, spurious three-digit hex codes in the middle of the content, as well as a lone "0" at the end. Since the downloaded content was javascript files, the extra chuck sizes in the middle of the content was causing the javascript to error out when loaded back to the browser.  I had a feeling it had something to do with gz compression or something else in the request and response headers, I just didn't know what!   Simply changing HTTP 1.1 to HTTP 1.0 solved it, and now my ESP32 code downloads are 100% clean.  I also couldn't find this HTTP 1.1 chunking problem with ESP32 mentioned anywhere else.   Thanks again.