esp8266/Arduino #3052: ESP-WROOM-2 better configuration

asetyde, 2017-03-16:
Hi , for CE certification i must change from NODEMCU to esp-wroom-2  , i want ask you a better choose in board manager.

From Espressif only FCC and CE certificated are ESP-07s , ESP12S , Wroom 2 , ESP12F

devyte, 2017-03-16:
I understand that nodemcu uses 12F,  and that is certified. Or do you have
an older nodemcu with 12E?

asetyde, 2017-03-16:
nodmecu uses 12E for standard , v3 as is called is not a nodemcu , i’ve last is not f . 

Espressif not speak about 12f only S

devyte, 2017-03-16:
This is news to me, I've bought several v3 (says lolin underneath), and
they look/feel/sound like nodemcu to me. Some friends alwoso bought
nodemcus, and they all came with 12F on board.
Do you mean that they are clones, and not the official one?

ha scritto:
>
> I understand that nodemcu uses 12F, and that is certified. Or do you have
> an older nodemcu with 12E?
>
>
> > Hi , for CE certification i must change from NODEMCU to esp-wroom-2 , i
> > want ask you a better choose in board manager.
> >
> > From Espressif only FCC and CE certificated are ESP-07s , ESP12S ,
Wroom 2
> > , ESP12F
auth/AQC6BvIBM9Ib1DueSQObbCC-xW1Bl2Hrks5rmOrYgaJpZM4Me82K>
https://github.com/esp8266/Arduino/issues/3052#issuecomment-287040086>, or
mute the thread <https://github.com/
AF4G6O77KoGEUtHZXZw3hTrX1Zx3Z4qyks5rmSixgaJpZM4Me82K>.

or mute
the thread

asetyde, 2017-03-16:
Yes , NODEMCU AMICA is original AI THINKER  with chip of usb cp2012 12E and 12E .
v3 has CH340 usb and now i see esp12f

good things if has certification !

asetyde, 2017-03-16:
http://www.smartarduino.com/the-newest-v3-nodemcu-based-on-esp-12f-from-esp8266_p94866.html

asetyde, 2017-03-16:
i see a "coming soon"

jjssoftware, 2017-03-16:
I see 415 Units in Stock Ships in 24hrs at http://www.smartarduino.com/the-newest-v3-nodemcu-based-on-esp-12f-from-esp8266_p94866.html

I've never actually bought one. I just tend to opt for the generic ESP12F module on it's own and put my own caps, resistors etc around it.

asetyde, 2017-03-17:
is not certificated at now , say doit/smart , will be soon

asetyde, 2017-03-17:
but about my first question , someone can reply me ?

asetyde, 2017-03-23:
@igrr  @Makuna  someone can help me ? I can go if I not use correct board in board managers , generic8285 is not correct for wroom 2 , 1mb but wroom has 2mb , also no description about pin

asetyde, 2017-03-23:
Also correct information on reference , because I think espduino wroom 2 is s2 , previous dev  board

asetyde, 2017-03-23:
https://github.com/esp8266/Arduino/issues/3074

closed 2017-03-23 by asetyde