esp8266/Arduino #3054: Station disconnects in a period in WIFI_AP_STA mode

vitotai, 2017-03-17:
### Basic Infos

#### Hardware
Hardware:			ESP-12 ( WEMOS D1 mini)
Core Version:      	2.3.0

### Description

The main issue is that the station disconnects from AP every 5 to 10 minutes when set to WIFI_AP_STA mode.

My real sketch creates a ESPAsyncWebServer and hosts a ServerSideEvent. It's funny that if a client, a browser, connects to it, then the disconnection won't happen.

A minor issue is that ESP8266 seems to connect to the saved AP before WiFi.begin() is called. My real sketch will open a configuration file before setup the network. The delay() in setup() is used to simulate it.

### Settings in IDE

Module:  			Core Development Module
Flash Size: 		 4MB/3MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:  40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include <ESP8266WiFi.h>

char *_apName="myapname";
char *_apPassword="myapname";

char *SSID="VH";
char *PASS="--------";

void setup()
{
	Serial.begin(115200);
	Serial.setDebugOutput(true);
	delay(2000);
	WiFi.mode(WIFI_AP_STA);
	WiFi.setAutoReconnect(false);
	Serial.printf("Start SoftAP\n");
	// start a AP.
	WiFi.softAP(_apName,_apPassword);

 	if( WiFi.status() == WL_CONNECTED){
 		Serial.printf("Already connected!!! who did it?\n");
 	}else{			
		WiFi.begin(SSID,PASS);
		bool timeout = false;
		int i=0;
		while(WiFi.status() != WL_CONNECTED && !timeout) {
        	delay(200);
    		timeout = i++ > 100;
   			Serial.printf(".");
    	}
 	}
 	// timeout or not
}

void loop()
{
	if(WiFi.status() != WL_CONNECTED) {
		Serial.printf("disconnected!!\n");
		WiFi.reconnect();
		bool timeout = false;
		int i=0;
		while(WiFi.status() != WL_CONNECTED && !timeout) {
        	delay(200);
    		timeout = i++ > 100;
   			Serial.printf(".");
    	}
		
	}
}
```

### Debug Messages

```
 1384, room 16 
tail 8
chksum scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 9
cnt 

connected with VH, channel 11
dhcp client start...
wifi evt: 0
ip:192.168.31.198,mask:255.255.255.0,gw:192.168.31.1
wifi evt: 3
Start SoftAP
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
Already connected!!! who did it?
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
state: 5 -> 2 (3c0)
rm 0
wifi evt: 1
STA disconnect: 3
```

Hagenmue, 2018-03-19:
Hello,
I have a similar problem. Have you solved the problem? If, so, please give a hint.

vitotai, 2018-03-20:
NO. I have no idea how to debug this issue.
The disconnection seems to happen only when there is NO connection of STATION IP, so my application seems to work fine.

Hagenmue, 2018-03-26:
Thank you for your your reply.