esp8266/Arduino #3059: Can't read or write SPI flash

kommancs96, 2017-03-18:
#### Hardware
Hardware:			ESP-12E (nodeMCU)

### Description

Can't use `ESP. flashRead()` and `ESP.flashWrite()`

### Settings in IDE

I use makefile and Atom.io as text editor

### Sketch

```cpp

#include <Arduino.h>

uint32_t* chunk = (uint32_t*)calloc(8, sizeof(uint32_t));

void testSPIFlash() {
  uint32_t from_addr = 0x100000;
  uint32_t to_addr = from_addr + 0xfb000;

  for (uint32_t i = from_addr; i < to_addr; i+=8) {
    uint32_t* data = (uint32_t*)malloc(8*sizeof(uint32_t));
    data[0]=0xfe3;
    data[1]=0xade57c;
    data[2]=0x102f;
    data[3]=0xb0fe3;
    data[4]=0xc0fe3;
    data[5]=0xf0e3;
    data[6]=0x1003;
    data[7]=0x8ead;
    ESP.flashWrite(i, chunk, 8*sizeof(uint32_t));
    
    delay(200);

    ESP.flashRead(i, chunk, 8*sizeof(uint32_t));
    Serial.print(i, HEX);
    Serial.print(":\t");

    //print all elements
    for (size_t i = 0; i < 8; i++) {
      Serial.print(chunk[i], HEX);
      Serial.print(" ");
    }

    Serial.println();
    delay(100);
  }
}

void setup() {
        delay(500);
        testSPIFlash();
}

void loop() {

}
```

### Debug Messages

As I run the codeabove, it prints constant zeroes, like this:

```
100000: 0 0 0 0 0 0 0 0
100008: 0 0 0 0 0 0 0 0
100010: 0 0 0 0 0 0 0 0
100018: 0 0 0 0 0 0 0 0
100020: 0 0 0 0 0 0 0 0
100028: 0 0 0 0 0 0 0 0
100030: 0 0 0 0 0 0 0 0
100038: 0 0 0 0 0 0 0 0
100040: 0 0 0 0 0 0 0 0
100048: 0 0 0 0 0 0 0 0
100050: 0 0 0 0 0 0 0 0
100058: 0 0 0 0 0 0 0 0
100060: 0 0 0 0 0 0 0 0
100068: 0 0 0 0 0 0 0 0
100070: 0 0 0 0 0 0 0 0
100078: 0 0 0 0 0 0 0 0
100080: 0 0 0 0 0 0 0 0
100088: 0 0 0 0 0 0 0 0
100090: 0 0 0 0 0 0 0 0
100098: 0 0 0 0 0 0 0 0
1000A0: 0 0 0 0 0 0 0 0
1000A8: 0 0 0 0 0 0 0 0
1000B0: 0 0 0 0 0 0 0 0
1000B8: 0 0 0 0 0 0 0 0
1000C0: 0 0 0 0 0 0 0 0
1000C8: 0 0 0 0 0 0 0 0
```

igrr, 2017-03-19:
You seem to be missing the step that flash sector must be erased (= set all bits to 1) prior to writing.

kommancs96, 2017-03-19:
@igrr I couldn't get it working, so I tried `CheckFlashConfig.ino` a bit modified:
```cpp
#ifndef DEBUG_LN(X, ...)
#define DEBUG_LN(X, ...) Serial.printf((X), ##__VA_ARGS__);\
Serial.println();
#endif

void setup(void) {
    Serial.begin(115200);
}

void loop() {

    uint32_t realSize = ESP.getFlashChipRealSize();
    uint32_t ideSize = ESP.getFlashChipSize();
    FlashMode_t ideMode = ESP.getFlashChipMode();

    DEBUG_LN("Flash real id:   %08X\n", ESP.getFlashChipId());
    DEBUG_LN("Flash real size: %u\n\n", realSize);

    DEBUG_LN("Flash ide  size: %u", ideSize);
    DEBUG_LN("Flash ide speed: %u", ESP.getFlashChipSpeed());
    DEBUG_LN("Flash ide mode:  %s", (ideMode == FM_QIO ? "QIO" : ideMode == FM_QOUT ? "QOUT" : ideMode == FM_DIO ? "DIO" : ideMode == FM_DOUT ? "DOUT" : "UNKNOWN"));

    if(ideSize != realSize) {
        DEBUG_LN("Flash Chip configuration wrong!");
    } else {
        DEBUG_LN("Flash Chip configuration ok.");
    }

    delay(5000);
}
```

The output I get is:
```shell
sum 0x2d
v00000000
~ld
   sytem param error, use last saved param!

Flash real id:   001640E0

Flash real size: 4194304

Flash ide  size: 524288
Flash ide speed: 40000000
Flash ide mode:  QIO
Flash Chip configuration wrong!
```
Does this mean that my flash chip is broken, or is this just a misconfiguration?
If yes, how can I properly reconfigure the board?

devyte, 2017-09-07:
The last message is indicative of a flash layout misconfiguration. I suggest trying with the Arduino IDE for testing, and reviewing your makefile build system, as well as your use of the flash tool, to figure out what's wrong.
This looks like user error in the make build system. Closing.

closed 2017-09-07 by devyte