esp8266/Arduino #3060: Strangest Thing Ever WEMOS D1 R2 Can't Upload

andrewqin47, 2017-03-18:
I am having probably the strangest issue with the Wemos D1 R2 Esp8266 Board that connects to an ultrasonic module to make a range finder.

Connection: (On-board)
D0 ---> RST

Connection: (Ultrasonic Sensor Module ---> Wemos D1 R2)
GND ---> GND
VCC ---> 5V
Trig ---> D6
Echo ---> D7

I first uploaded a sketch from my laptop PC to see if the ultrasonic module can measure distance and it worked.

Then, I connect this board to my desktop PC, selected the right port and board everything, and tried to upload another sketch, this time, I got this error message:

Arduino: 1.8.1 (Windows Store 1.8.1.0) (Windows 10), Board: "WeMos D1 R2 & mini, 80 MHz, 921600, 4M (3M SPIFFS)"

Sketch uses 236381 bytes (22%) of program storage space. Maximum is 1044464 bytes.
Global variables use 34332 bytes (41%) of dynamic memory, leaving 47588 bytes for local variables. Maximum is 81920 bytes.
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

I thought: "OK, maybe I should upload it using the laptop still?" Therefore I switched back to my laptop, but the same error message occurs when uploading.

Isn't that strange??? I uploaded the first time perfectly, then I could not upload sketches using a different computer, and the error wouldn't be fixed after switching back to the original computer.

Please help me out, I browsed through many forums and couldn't find any cases similar to mine, and I am very frustrated.

izzandienurrahman, 2017-03-26:
try change the upload speed to 115200

andrewqin47, 2017-03-26:
I tried it but it did not help.

closed 2017-09-07 by devyte