esp8266/Arduino #3061: SDWebServer SD.h error

SiteMasterDK, 2017-03-18:
Arduino ide 1.8.1
esp8266 2.3.0

i have just reinstalled arduino ide and esp8266 2.3.0

i get this error when i try to use my esp8266 and a SD card reader

In file included from D:\Dokumenter\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,

                 from D:\Dokumenter\Arduino\libraries\SD\src/utility/SdFat.h:29,

                 from D:\Dokumenter\Arduino\libraries\SD\src/SD.h:20,

                 from C:\Users\SiteMaster\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WebServer\examples\SDWebServer\SDWebServer.ino:35:

D:\Dokumenter\Arduino\libraries\SD\src/utility/Sd2PinMap.h:510:2: error: #error Architecture or board not supported.

 #error Architecture or board not supported.

  ^

Multiple libraries were found for "SD.h"
 Used: D:\Dokumenter\Arduino\libraries\SD
 Not used: C:\Users\SiteMaster\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
Error compiling for board ESPino (ESP-12 Module).

`
/*
  SDWebServer - Example WebServer with SD Card backend for esp8266

  Copyright (c) 2015 Hristo Gochkov. All rights reserved.
  This file is part of the ESP8266WebServer library for Arduino environment.

  This library is free software; you can redistribute it and/or
  modify it under the terms of the GNU Lesser General Public
  License as published by the Free Software Foundation; either
  version 2.1 of the License, or (at your option) any later version.

  This library is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the GNU
  Lesser General Public License for more details.

  You should have received a copy of the GNU Lesser General Public
  License along with this library; if not, write to the Free Software
  Foundation, Inc., 51 Franklin St, Fifth Floor, Boston, MA  02110-1301  USA

  Have a FAT Formatted SD Card connected to the SPI port of the ESP8266
  The web root is the SD Card root folder
  File extensions with more than 3 charecters are not supported by the SD Library
  File Names longer than 8 charecters will be truncated by the SD library, so keep filenames shorter
  index.htm is the default index (works on subfolders as well)

  upload the contents of SdRoot to the root of the SDcard and access the editor by going to http://esp8266sd.local/edit

*/
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <SPI.h>
#include <SD.h>

#define DBG_OUTPUT_PORT Serial

const char* ssid = "**********";
const char* password = "**********";
const char* host = "esp8266sd";
[... 202 lines cut ...]
    DBG_OUTPUT_PORT.println("MDNS responder started");
    DBG_OUTPUT_PORT.print("You can now connect to http://");
    DBG_OUTPUT_PORT.print(host);
    DBG_OUTPUT_PORT.println(".local");
  }

  server.on("/list", HTTP_GET, printDirectory);
  server.on("/edit", HTTP_DELETE, handleDelete);
  server.on("/edit", HTTP_PUT, handleCreate);
  server.on("/edit", HTTP_POST, [](){ returnOK(); }, handleFileUpload);
  server.onNotFound(handleNotFound);

  server.begin();
  DBG_OUTPUT_PORT.println("HTTP server started");

  if (SD.begin(SS)){
     DBG_OUTPUT_PORT.println("SD Card initialized.");
     hasSD = true;
  }
}

void loop(void){
  server.handleClient();
}
`

DarkBeardRUS, 2017-03-24:
Hi!
And for what device do you compile this source?

SiteMasterDK, 2017-03-24:
have tried with
NodeMCU 1.0
Generic ESP8260 module
ESPino

devyte, 2017-09-07:
Please retry with latest git. I'm not aware of any current compiler issues with SD.

devyte, 2017-10-10:
No feedback in over a month, closing.

closed 2017-10-10 by devyte