esp8266/Arduino #3062: How do I read a file from SPIFFS into a variable for HTTP POST?

Daemach, 2017-03-19:
I'm trying to send local log files from the ESP8266 SPIFFS to a web server.  The following code is not working.  How do I read a file from SPIFFS into an HTTP POST?

`  File f = SPIFFS.open("/debugLog.txt", "r");
      if (!f) {
          Serial.println("file open failed");
      } 

   // This dumps the log data to the serial port properly
    while (f.available()){
      Serial.println(f.readStringUntil('\n'));
    }

    String debugURL = debugLogURL += MAC;
    String debugLogData;

    //This doesn't work...
    while (f.available()){
      debugLogData += f.readString();
    }

    Serial.println(debugLogData);  //Drops an empty string into serial

    http.begin(debugURL, fingerprint);
    http.addHeader("Content-Type", "application/x-www-form-urlencoded");
    http.POST("logData=" + debugLogData);    //Is this the correct format?
    String response = http.getString();
    response.trim();
    Serial.println(response);
    http.writeToStream(&Serial);
    http.end();`

JamesGKent, 2017-03-20:
well it looks to me like you are opening the file and reading it when you dump to serial, you need to either close the file and open it again or seek back to the beginning of the file before you can read again. or perhaps a better approach would be to read only once and send it both to serial and your string in the same loop.

adjavaherian, 2017-03-22:
I basically want to do the same thing but with a larger image file.  Do I need break the image up into bytes and POST each byte?

Daemach, 2017-03-22:
Thanks James.  Here's what ended up working:

```
SPIFFS.begin();

  if (!SPIFFS.exists("/formatComplete.txt")) {
    Serial.println("Please wait 30 secs for SPIFFS to be formatted");
    SPIFFS.format();
    Serial.println("Spiffs formatted");
    
    File f = SPIFFS.open("/formatComplete.txt", "w");
    if (!f) {
        Serial.println("file open failed");
    } else {
        f.println("Format Complete");
    }
  } else {
    Serial.println("SPIFFS is formatted. Moving along...");
  }

  if (SPIFFS.exists("/debugLog.txt")){

      File f = SPIFFS.open("/debugLog.txt", "r");
      if (f && f.size()) {
          Serial.println("Dumping log file");
                
          String debugURL = debugLogURL += MAC;
          String debugLogData;
      
          while (f.available()){
            debugLogData += char(f.read());
          }
          f.close();
               
          Serial.println("=====================================");
          Serial.println(debugLogData);
          Serial.println("=====================================");
    
          http.begin(debugURL, fingerprint);
          http.addHeader("Content-Type", "application/x-www-form-urlencoded");
          int httpCode = http.POST("logData=" + debugLogData);
    //    String response = http.getString();
    //    response.trim();
          Serial.print("Return code = ");  Serial.println(httpCode);
          http.writeToStream(&Serial);
          http.end();

          if (httpCode == 200){
              //Transmitted successfully.  New log.
            Serial.println("Log transmitted successfully.  Truncating...");
            debugLog = SPIFFS.open("/debugLog.txt", "w");
            if (!debugLog) {
                Serial.println("Opening file for write failed");
            }
            debugLog.close();
          } else {
              //Error of some kind - keep the log and append.
              Serial.println("Submission unsuccessful - appending to existing log...");

          }         
      } else {
          Serial.println("Debug log is empty");
      }
      
  }
  SPIFFS.end();
```

devyte, 2017-09-07:
Given the last comment, closing.

closed 2017-09-07 by devyte