esp8266/Arduino #3064: Error resolving.

Joypacharya, 2017-03-20:
i am using esp8266 nodemcu V 1.0 and i sat out everything. Still i found unknown error at running time and my device has been restarted.

See this issues and tell me what kind of error is this,
   
Exception (28):
epc1=0x4023571f epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: sys 
sp: 3ffffd20 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000012 40211f98 3ffeef58 00000001  
3ffffee0:  ffffffff 3ffe9bb4 3ffeef58 00000000  
3ffffef0:  00000000 40201ff1 00000000 00000000  
3fffff00:  f7ffffff 60000600 3ffe9ed1 60000600  
3fffff10:  00000007 00000007 00000001 40239544  
3fffff20:  00000008 00000000 00000008 4023956e  
3fffff30:  40211ab5 3ffee020 3ffeef80 01a378b3  
3fffff40:  3fff6994 00001000 40211efe 00000008  
3fffff50:  3ffeef80 3ffee0d4 40211fab 3ffee0d4  
3fffff60:  3ffeef80 0082f0c4 3ffeef80 60000600  
3fffff70:  40223685 3ffee0d4 3ffeef80 01a37216  
3fffff80:  402236ca 3fffdab0 00000000 3fffdcb0  
3fffff90:  3ffeef90 00000000 40000f65 3fffdab0  
3fffffa0:  40000f49 0000a199 3fffdab0 40000f49  
<<<stack<<<

devyte, 2017-03-20:
There is a tool for the Arduino IDE called EspExceptionDecoder. I suggest
installing it and giving it a try.

Joypacharya, 2017-03-20:
I already done this but i could not find proper error in my code. You just put this stack on your exception decoder and tell me.

Joypacharya, 2017-03-21:
Decoding 13 results
0x402356e7: register_chipv6_phy_init_param at ?? line ?
0x4020f94a: pp_attach at ?? line ?
0x40201ff1: __wrap_register_chipv6_phy at C:\Users\JOY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_phy.c line 261
0x40239508: pm_wakeup_init at ?? line ?
0x40239532: pm_wakeup_init at ?? line ?
0x40211a7d: pm_set_sleep_time at ?? line ?
0x40211ec6: pm_get_sleep_type at ?? line ?
0x40223692: ets_timer_handler_isr at ?? line ?
0x402130c3: pm_post at ?? line ?
0x4020cbaf: loop_task at C:\Users\JOY\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 57

lmk1988, 2017-03-22:
Seems like a timer issue.
Here are some possible reason:
- Used delay() in timer triggered function
- Performing task that are too long in timer

I might be wrong on this though.
Could try moving your codes into loop() and use a millisecond variable to track when to run the code

devyte, 2017-09-08:
@Joypacharya what is the complete minimal sketch to reproduce the issue? I see from your decoded stack that you are using core 2.3.0. Does it also happen with latest git?

devyte, 2017-10-10:
Requested sketch, no feedback in over a month. Closing.

closed 2017-10-10 by devyte