esp8266/Arduino #3066: Erase saved WiFi access points

hieuvt6-fti, 2017-03-21:
I used `WiFiMulti` in the past to connect to different WiFi networks. Now I want to forget all of those WiFi configurations (the module keep trying to connect to a (now turned) bad AP).
I tried to follow #1494 and investigate these solutions:

`
  ESP.eraseConfig();
  *((int*)0) = 0;
`

`
  ESP.eraseConfig();
  ESP.reset();
`

Neither worked.

Do anyone know any better way to do this?

tablatronix, 2017-03-26:
in code or permanently ?
You can use the esptool to erase_flash entirely.

hieuvt6-fti, 2017-03-27:
Ah, yes. after a while I decided to erase flash and it turned out to be the solution.

But now I have two questions.

1. I read some GitHub issues and learned that WiFiMulti is not actually using the flash space reserved for storing WiFi credentials. If so, how is it storing WiFi credentials in a non-volatile way?

2. If I want to erase WiFiMulti's stored credentials during run time, what are my options?

tablatronix, 2017-03-27:
It stores the last successful ap but not all.
You can disable persistent to disaae autonsaving

devyte, 2017-09-08:
@hieuv did calling persistent(false) address your issue?

hieuv, 2017-09-09:
To be honest with you, this was so long ago that I do not remember any details anymore. I will see if I can reproduce the issue and see if the solution above fixes it.

devyte, 2017-10-10:
@hieuv were you able to reproduce the issue?

barsznica, 2018-01-02:
WiFi.disconect() worked for me first go.

ESP.eraseConfig then ESP.reset() OR  ESP.eraseConfig() then ESP.restart() did not work for me.
I did not try the `*((int*) 0) = 0;` thing.

The answer to this question was covered [here](url=https://github.com/tzapu/WiFiManager/issues/142) and [here](url=https://github.com/tzapu/WiFiManager/issues/142).

devyte, 2018-02-28:
Closing due to age and previous comment.

closed 2018-02-28 by devyte