esp8266/Arduino #3069: ESP-01 and Arduino Mega Setup.

ongvanor, 2017-03-22:
### Basic Infos

#### Hardware
Hardware: ESP-01 
http://www.bitstoc.com/index.php?route=product/product&path=89&product_id=121
Core Version: Firmware 1.1.2 updated it using this 
http://www.espruino.com/files/ESP8266_AT25-SDK112-512k.bin

### Description

How to connect esp-01 to arduino mega using this library. hardware or software serial? 
I want to try this library, what i was using before is the WiFiEsp Library of https://github.com/bportaluri/WiFiEsp and PubSubClient for MQTT connection,
there are times that i cant receive the message from the subscribe topic sometimes  receive.
so i wanted to try this. 

i tried using ESP-12E http://www.bitstoc.com/index.php?route=product/product&path=89&product_id=172

but i need more pins for my project. so i planned to use arduino mega and a esp-01 module

### Settings in IDE

No idea for now

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

### Debug Messages

evbelda, 2017-04-17:
I tried to use the "Generic ESP8266 Module" with Mega and it doesn't accept the config.
So, I have the same problem. I need this library to use an SSL Client socket.
Any idea?

igrr, 2017-05-15:
This project enables programming of the ESP8266 itself, not controlling it using an external Arduino MCU.

closed 2017-05-15 by igrr