esp8266/Arduino #3070: Function to change ESP8266WebServer port

mcteusz, 2017-03-22:
Hello
I am new in github and i am not sure if this is correct place to ask for help but i desperately need possibility to set port for ESP8266WebServer object.
I have tried to use it by pointer but no way.
Maybe somone could give me advice or can write simple function like ESP8266WebServer.setPort(int port)

Regards

krzychb, 2017-03-22:
Hi @mcteusz,

https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WebServer/src/ESP8266WebServer.h#L66-L67

mcteusz, 2017-03-22:
Can it change port of previoulsy declared ESP8266WebServer variable?
I thought this is only constructor.
Anyway when my code looks like that:

`#define WEBSERVER_PORT      8080

ESP8266WebServer WebServer(WEBSERVER_PORT);

struct MyStruct
{
  int           WebserverPort;
  (...)
} Settings;

WebServer(Settings.WebserverPort);`

And i have error:

> no match for call to '(ESP8266WebServer) (int&)'

Just why?

krzychb, 2017-03-23:
Yes, ESP8266WebServer is a constructor.
In ESP8266WebServer.h do not see any method exposed to change the port after ESP8266WebServer object is created.
So the answer is you cannot change the port of already created ESP8266WebServer object

mcteusz, 2017-03-29:
So is it possible to improve library? Unfortunally i am not feeling up to.

devyte, 2017-09-08:
You can create and destroy the object dynamically:
ESP8266Webserver *server = new ESP8266Webserver(80);

...

if (neednewport)
{
  delete server;
  server = new ESP8266Webserver(newport);
}

Not a perfect solution, but it is valid, it works with the current code base, and it doesn't require a code change in the core.

Closing.

closed 2017-09-08 by devyte