esp8266/Arduino #3071: Wire.begin(0, 5); does not work

danriches, 2017-03-22:
### Basic Infos
I'm using several libraries which I've removed for testing and I've checked the hardware works on the designated i2c bus pins which it does. However I'm using the standard i2c pins for other uses and need to reroute to using pins 0 and 5 for i2c usage. I adjusted the adafruit library so that readCoefficients is public so I don't call the begin function and stuff up the i2c config.

Any ideas?

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description
I2C does not work on pins 0 and 5

### Settings in IDE
Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#include <Arduino.h>
#include <Wire.h>
#include <Adafruit_MPL115A2.h>

Adafruit_MPL115A2 mpl115a2;
int Humidity = 0;
float Temp = 0.0;
bool validReading = false;
float pressureKPA = 0, temperatureC = 0;

void GetTemp()
{
  validReading = false;
  Wire.beginTransmission(0x28);
  Wire.requestFrom(0x28, 4);

  if(Wire.available() == 4)
  {
    Serial.println("Reading Sensor values...");

    int b1 = Wire.read();
    int b2 = Wire.read();
    int b3 = Wire.read();
    int b4 = Wire.read();

    Wire.endTransmission();

    int rawHumidity = b1 << 8 | b2;
    rawHumidity = (rawHumidity &= 0x3FFF);
    Humidity = (int)(100.0 / pow(2,14) * rawHumidity);

    b4 = (b4 >> 2);
    int rawTemp = b3 << 6 | b4;
    Temp = (165.0 / pow(2,14) * rawTemp - 40);

    Serial.println("TEMP: " + String(Temp));
    Serial.println("HUMIDITY: " + String(Humidity));

    validReading = true;
  }
  delay(1);
}

void GetPressure()
{
  mpl115a2.getPT(&pressureKPA,&temperatureC);
  Serial.print("Pressure (kPa): "); Serial.print(pressureKPA, 4); Serial.print(" kPa  ");
  Serial.print("Temp (*C): "); Serial.print(temperatureC, 1); Serial.println(" *C both measured together");
  pressureKPA = mpl115a2.getPressure();
  Serial.print("Pressure (kPa): "); Serial.print(pressureKPA, 4); Serial.println(" kPa");
  temperatureC = mpl115a2.getTemperature();
  Serial.print("Temp (*C): "); Serial.print(temperatureC, 1); Serial.println(" *C");
  delay(1);
}

void setup() 
{
  // put your setup code here, to run once:
  delay(1000);
  Wire.begin(0, 5);
  mpl115a2.readCoefficients();
  Serial.begin(115200);
}

void loop() 
{
  // put your main code here, to run repeatedly:
  GetTemp();
  GetPressure();
  delay(1000);

}

### Serial output
Pressure (kPa): 309.7104 kPa  Temp (*C): -73.1 *C both measured together
Pressure (kPa): 309.7104 kPa
Temp (*C): -73.1 *C

tablatronix, 2017-03-26:
Have you tried this with 2 different arbitrary pins? Or is this specific to 0,5 ?

danriches, 2017-03-29:
I've tried a whole bunch of different pins and it seems that I can't use pin 0 for anything at all. I've now reconnected the ILI9341 display and the touchscreen and that won't work using it either and this is on two different ESP8266-12 modules. I really do need this to work and pin 0 is the only one left. Anyone got any ideas???

Thanks.

cbrum11, 2017-03-29:
So I'm not sure if this is related or not, but if you check out the WEMOS page it states

> All of the IO pins have interrupt/pwm/I2C/one-wire support **except D0**

 ...and D0 is GIPO Pin 0 I believe.

https://www.wemos.cc/product/d1-mini-pro.html

Good luck
-Chase

danriches, 2017-03-29:
One test I've just done is to set pin 16 as an input and pin 0 as an output and each time around the loop function I toggle pin 0. I've also set a delay at the start so I can see when it happens and all I get is one transition from high to low and then it latches at low. There is definitely something not right regarding pin 0 here and I've checked with a scope too. Grrr, this is driving me nuts!!

I've also tried pin 0 as an input and manually driving it high and low and again I only get a reading of high initially and then low once driven low, then when driven high again it still reads low.

I then changed to a completely different module than the other two, it's an 07 series and that works fine. I tried all sorts of flash size configuration settings with the other two and that didn't change it either. Very strange, but at least I can use an 07 for this project and although it has less flash (1Mb not 4Mb) it should still work as long as I have enough memory?!?!

cbrum11, 2017-03-29:
I don't know if this is practical with your current setup, but have you considered trying a pull-up resistor on pin 0?  I know that pin is used in boot-loader mode when pulled low.

Also... this was interesting

> Here’s what we found: 40ms after startup, the GPIO0 line is driven with a signal at around 350 Hz for around 100 ms. So make sure you don’t rely on GPIO0 being stable for the first ~200 ms after startup.

https://zoetrope.io/tech-blog/esp8266-bootloader-modes-and-gpio-state-startup

danriches, 2017-03-29:
Hi all, 

After checking everything again and again, ad infinitum (almost), I've finally found the issue with pin 0. Using some freezer spray I managed to locate a hairline fracture on the carrier board that the ESP is sitting on and yes, you guessed it, pin 0 had the fracture. The act of pressing the upload button caused enough flex to connect the pin so I can upload new code. Then afterwards the code would run and the board slowly flexed back to disconnect the trace. I'm now going to dig out the soldering iron and reflow every pin to ensure this is fixed.

Thanks to everyone that replied, it's much appreciated!!!

Dan

tablatronix, 2017-03-29:
I would avoid gpio 0 if you can, it outputs clk signal on boot could cause noise in your bus

danriches, 2017-03-30:
There being a bit of noise initially on GPIO 0 is not a real problem, but one issue is that after changing the pin assignments around and trying to use pin 16 with I2C gives another issue. The pin is just held high, however if I disable I2C and set pin 16 to an output and toggle it then I get a toggling output. So it seems that the I2C (Wire) library doesn't work with GPIO 16 as either the clock or data pin. 

Can anyone confirm this?

danriches, 2017-03-31:
Well it seems as though the Wire library won't work with pin 16 and the UTFT library can't use pin 16 either. Luckily the touchscreen library can use pin 16 for the CS and that means I can switch everything to use the standard i2c pins (4, 5). Thank goodness for that!

You can close the issue if you like although I could never get i2c to work with pin 0 or 16. I guess it's always best to use the dedicated pins for the job!!

Thanks all for your input, much appreciated ;)

earlephilhower, 2019-07-25:
Closing.  GPIO16 comes from the RTC and is not supported by Wire or many other speed-critical libraries (it's slower since it's not off the main core and requires writing different registers than the other 16 IOs).

closed 2019-07-25 by earlephilhower