esp8266/Arduino #3072: Can't re-enable WiFi after ESP.deepSleep(uS, WAKE_RF_DISABLED) ?

sheffieldnick, 2017-03-22:
### Basic Infos

#### Hardware
Hardware:			WeMos D1 mini
Core Version:      	v2.2.0

### Description

My sketch spends most of its time in deep sleep, just using the WiFi when it first powers up. This works fine from a full power-cycle, but not if is reset (reset button, or new sketch uploaded over serial). In that case I can't seem to switch the WiFi back on? I presume the RF is disabled from the deep sleep, even through a reset, and WiFi.mode(WIFI_STA) isn't enough to switch it back on?

I get:

`mode : sta(MAC)
add if0
f r0, 
Connecting to WLAN_SSID
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!`
....
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
scandone
reconnect
f 0, error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
....
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
scandone
reconnect
f -180, error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
....
error: pll_cal exceeds 2ms!!!
error: pll_cal exceeds 2ms!!!
scandone
del if0
usl
mode : null`

### Settings in IDE

Module:  			WeMos D1 R2 & mini
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		SERIAL
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <ESP8266WiFi.h>
extern "C" {
#include <user_interface.h>
}

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  // Connect to WiFi if we have just powered on (*not* if waking from deep sleep)
  rst_info *resetInfo = ESP.getResetInfoPtr();
  if (resetInfo->reason != REASON_DEEP_SLEEP_AWAKE) {
    WiFi.mode(WIFI_STA);
    WiFi.begin(WLAN_SSID, WLAN_PWD);
    WiFi.setAutoConnect(false);
    WiFi.setAutoReconnect(false);
    WiFi.waitForConnectResult();
    // do some networky stuff

    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
  }
}

void loop() {
  // do regular non-networking stuff

  // enter deep sleep, with radio off upon waking
  unsigned int sleep_secs = 10;
  ESP.deepSleep(sleep_secs*1000000, WAKE_RF_DISABLED);
}
```

NB: I'm using WiFi.setAutoConnect(false)/setAutoReconnect because otherwise the board was always trying to connect to the WiFi immediately upon waking from deep sleep, even if WiFi.disconnect(), WiFi.mode(WIFI_OFF) and WAKE_RF_DISABLED had been called.

sheffieldnick, 2017-03-22:
#644 mentions using:
ESP.deepSleep(1, WAKE_RF_DEFAULT)
as a hack to switch the RF back on, but surely there is a better way?

igrr, 2017-03-23:
No, unfortunately there isn't. If you ask it to do a deep sleep reset and wake up with RF disabled, RF is going to be exactly that — disabled.

sheffieldnick, 2017-03-23:
OK, is there a fast way to test the RF disabled/enabled state? So my code can distinguish between my 'normal' REASON_DEEP_SLEEP_AWAKE, and when it comes awake from a brief deep sleep used to enable the RF?

igrr, 2017-03-23:
Writing a flag into RTC memory prior to going into deep sleep is probably the easiest way. ESP8266 SDK doesn't provide a function to query RF state.

sheffieldnick, 2017-03-23:
Thanks, that seems to be working nicely.

sheffieldnick, 2017-03-28:
Just noticed this line in Esp.h:

`#define RF_MODE(mode) int __get_rf_mode() { return mode; }`

can't I use __get_rf_mode() to read the current RF state?

jeanfbrito, 2017-03-30:
This is what I'm searching for. I was using LUA scripts on Nodemcu, but it eats alot of battery.
Now I will program in "Arduino" and need to get if the RF is ON, to then send the packets.

JamesNewton, 2017-10-04:
How does one call __get_rf_mode() in an ino file?

pfabri, 2017-12-01:
@JamesNewton 

Not sure about a .ino file. In a .cpp file you could do this: `extern int __get_rf_mode();`. This will then give you access to `__get_rf_mode()`, . You could then do `int mode = __get_rf_mode()`but according to my test this always returns `-1` regardless of the active RF_MODE. Sad, I know.

pfabri, 2017-12-02:
I've discovered a **workable solution** in a [gist](https://gist.github.com/chaeplin/be4d9ca41a991aa327bb)

This is all you need to be able to toggle your WiFi chip after `ESP.deepsleep()` calls. Both functions work before and after any of these deepsleep invokations: `WAKE_RF_DEFAULT`, `WAKE_RF_DISABLED`, and probably the other two too, (`WAKE_RFCAL`, `WAKE_NO_RFCAL`), but these I have not tested.

	extern "C" {
		#include "user_interface.h"  // Required for wifi_station_connect() to work
	}

	#include <Arduino.h>
	#include <ESP8266WiFi.h>

	#define FPM_SLEEP_MAX_TIME 0xFFFFFFF

	void WiFiOn();
	void WiFiOff();

	//------------------------------------------------------------------------------

	void setup() {

		WiFiOn();
		delay(5000);
		WifiOff();
		delay(5000);
		ESP.deepSleep(1e6 * 10, WAKE_RF_DISABLED); // sleep 10 seconds
	
	}

	void loop() {
		;
	}
	//------------------------------------------------------------------------------

	void WiFiOn() {

		wifi_fpm_do_wakeup();
		wifi_fpm_close();

		//Serial.println("Reconnecting");
		wifi_set_opmode(STATION_MODE);
		wifi_station_connect();
		//wifi_connect();
	}

	void WiFiOff() {

		//Serial.println("diconnecting client and wifi");
		//client.disconnect();
		wifi_station_disconnect();
		wifi_set_opmode(NULL_MODE);
		wifi_set_sleep_type(MODEM_SLEEP_T);
		wifi_fpm_open();
		wifi_fpm_do_sleep(FPM_SLEEP_MAX_TIME);

	}

2000jago, 2017-12-16:
The code above does not compile verbatim.  Arduino IDE complains about line 19 "'WifiOff' was not declared in this scope"...

pfabri, 2017-12-16:
@2000jago Yup, I wrote `WifiOff` instead of  `WiFiOff` by mistake. Corrected.

2000jago, 2017-12-16:
@pfabri I can confirm (on wemos D1 mini) this does indeed work and it works well.  Thank you sir!

torntrousers, 2017-12-17:
Would be nice to add those functions to the core ESP.h class

angeloh1969, 2018-03-28:
else
      {line = ("Conexion OK a WiFi");
        Serial.println (line);
       
      }
  return;
}

mahesh2000, 2018-07-13:
hi, has anyone tested the power usage while in sleep mode? i'm getting 3mA and can't go lower. 3mA is too much for my purpose. thanks!

ibaranov-cp, 2018-07-30:
Does your module have an LED? is it on? If so, that's likely your power drain.

Tech-TX, 2020-02-03:
Regarding the original post, I tried every trick I could think of, and I have to agree with **igrr** about whether you can re-enable WiFi after ESP.deepSleep(time, WAKE_RF_DISABLED)

> No, unfortunately there isn't. If you ask it to do a deep sleep reset and wake up with RF disabled, RF is going to be exactly that — disabled.

devyte, 2020-02-03:
Closing in view of previous comment.

closed 2020-02-03 by devyte

amadeuspzs, 2020-05-27:
I have spent many many hours trying to fix this until I came across this (closed) issue.

Please @igrr and team can you add to the documentation a warning about using WAKE_RF_DISABLED and the tip about using WAKE_RF_DEFAULT to get things up and running again?

devyte, 2020-05-27:
@amadeuspzs Please make your proposal in a PR.

Cicicok, 2020-12-09:
Actually I found another solution, which I've successfully tested. The trick is after waking from `WAKE_RF_DISABLED` the radio has to be put into sleep and immediatelly woken up.

```
WiFi.forceSleepBegin();
delay(1);
WiFi.forceSleepWake();
delay(1);
```

Source: https://www.bakke.online/index.php/2017/05/21/reducing-wifi-power-consumption-on-esp8266-part-2/

Tech-TX, 2020-12-16:
Yep, works fine to wake the modem after ESP.deepSleep(timeus, WAKE_RF_DISABLED), thanks for finding that!  It boots at 38mA for about 1.5ms, then drops to 18mA and stays there until you enable the modem.

[image]

I'll do a PR to add it to the docs.

I've a suspicion @torntrousers might find this of interest.  ;-)

panosss, 2022-01-04:
> Actually I found another solution, which I've successfully tested. The trick is after waking from `WAKE_RF_DISABLED` the radio has to be put into sleep and immediatelly woken up.
>
> ``` [...]

Does not work for me, maybe something has changed in the code?

nbolton, 2022-12-03:
> Does not work for me

Seems to still work for me. Maybe we have different versions?

nbolton, 2022-12-03:
Hmm, or maybe not. It's a bit unpredictable. Sometimes after using the workaround (below), the ESP8266 will consume 40 mA (not the expected 70 mA), the connection will fail.

```
WiFi.forceSleepBegin();
delay(1);
WiFi.forceSleepWake();
delay(1);
```

After it gets in this state, the only way to fix it is to hard power cycle the ESP8266, or call `ESP.deepSleep(n, WAKE_RF_DEFAULT)`

DavvidU, 2025-09-18:
> Hmm, or maybe not. It's a bit unpredictable. 

I don't know Your specific scenario or wether this will help You, but I'm able to wake up radio like this:

`WiFi.forceSleepWake();
  delay(10);
  WiFi.persistent(false);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);`

after 

`ESP.deepSleep(sleepTime_us, WAKE_RF_DISABLED);`

if in the previous cycle I had also done this wake up. After I hit a cycle where I'm not calling the wake up code, in the next cycle the same wake up code won't work anymore.