esp8266/Arduino #3080: Blocking HTTPUpdater under certain conditions

WayneManion, 2017-03-24:
I would like to be able to use HTTPUpdate, but I don't want just anyone with network access to be able to upload new firmware to my device. How can I keep the ESP8266 from serving up "/update" under certain conditions?

I've tried things like setting httpServer.on("/update", serveSomeOtherPage) to override the ESP8266HTTPUpdateServer, but that didn't work. The only way I can keep the ESP8266 from serving "/update" is by never loading httpUpdater.setup(httpServer) in the first place. 

What I would like to do is run the ESP8266 in station mode 99% of the time. When an http client make a certain HTTP get request, it triggers the ESP to start a soft AP. I then want the /update page accessible ONLY to access point clients (ie, 192.168.4.x addresses). This way the critical transmissions are at least encrypted via WPA2 and can only be performed by those who know the password to the soft AP. 

A second option is for a particular MQTT message to trigger the whole "launch soft AP" scheme. 

Is there anything I can do to tell an ESP8266HTTPUpdateServer to stop listening short of an ESP.restart()?

I am aware that authentication is an option, but without encryption, it isn't enough to make me happy.

jjssoftware, 2017-03-27:
implement your own authentication scheme and implement TLS encryption: 

* https://jjssoftware.github.io/secure-your-esp8266/
* https://jjssoftware.github.io/esp8266-password-complexify/
* https://jjssoftware.github.io/esp8266-rfc6238-totp/

WayneManion, 2017-03-27:
Those are all good suggestions. I am looking to prevent possible intruders on my local network from being able to access the HTTPUpdate page.

davisonja, 2017-03-27:
Is switching to an ESP-pulls update approach on the cards?
If MQTT is in scope...

jjssoftware, 2017-03-28:
You could come up with some sort of switching mechanism into AP mode but if you have a problem with intruders on your network, what's to stop these intruders getting access to the device when it's in AP mode?

To my mind, local network or public network doesn't really matter - the problem is the same: you want to prevent access to some sensitive resource / feature on your widget that hosts it's own web server. This really isn't anything unusual in the realm of web applications (not just ESP8266 devices) and there's generally only one universally accepted way of implementing this type of protection - implementation of an authentication mechanism which is usually also coupled with role based or user based access control based on sessions / a session token (cookie).

In short if you're stuck on HTTP, you need to solve the problem with a HTTP type of solution..implement an authentication scheme and the best you can do is do what you can to prevent "intruders" getting unauthenticated direct access to your device web server.

I do agree that it can be a problem that ESP8266 devices don't have the power / required available RAM to host their own HTTP TLS stack (aside that we don't actually have a framework to enable it). Add to that the fact that ESP devices are solely WiFi devices and this can be a challenge.

If you can augment this setup with MQTT or some other protocol then the suggestion by @davisonja could have some mileage. But whatever you do, if you have potential intruders on the LAN..you're going to need to do something pretty clever.

devyte, 2017-09-07:
What you need is webserver simple authentication. There is an example available.
This is not the right place to ask for assistance for your particular project. This is an issue tracker, meant for tracking issues in the core. Please refer to esp8266.com or stackoverflow for assistance.

closed 2017-09-07 by devyte