esp8266/Arduino #3082: How do you see what functions are in the .text RAM segment?

ericmnel, 2017-03-24:
### Basic Info

#### Hardware

Hardware:	Wroom-02
Core Version: 2.3.0

### Description

In this link - 

http://tech.scargill.net/esp8266-ram/

It talks about the .text section of memory, where time critical code is loaded into memory at bootup.   Is there a way to see what functions are in that segment, and is there a way to mark functions to not be loaded in memory at bootup?

### Settings in IDE

Module: Wroom-02
Flash Size: 4MB/1MB
CPU Frequency:	80Mhz
Flash Mode:	dio
Flash Frequency:	40Mhz
Upload Using:	SERIAL
Reset Method:	ck

jjssoftware, 2017-03-29:
I'm not aware of any way to list all functions that are setup to be located in .irom0.text after the fact / following compile. Calling `xtensa-lx106-elf-size -A "/path/to/your-compiled-elf-file.elf"` will give you an overview of sizes and addresses for all the various sections / types of memory..not sure if that's good enough for you though.

> is there a way to mark functions to not be loaded in memory at bootup?

Generally all functions are located in precious RAM by default unless you decide to attribute the function with `ICACHE_FLASH_ATTR` and if you do this, this indicates that the function is to be located in .irom0.text flash and it will be copied to RAM as required at runtime when necessary.

davisonja, 2017-03-29:
Will a linker .map file help?
-M--print-mapPrint a link map to the standard output. A link map provides
information about the link, including the following:

   - Where object files and symbols are mapped into memory.
   - How common symbols are allocated.
   - All archive members included in the link, with a mention of the symbol
   which caused the archive member to be brought in.

-Map mapfilePrint a link map to the file mapfile. See the description of
the `-M' option, above.

closed 2017-03-31 by ericmnel