esp8266/Arduino #3083: Arduino OTA tcp based or UDP based ?

Joypacharya, 2017-03-25:
Hello everyone,
Actually i am working on one ESP project. I am using nodemcu v 1.0.
I am upload file via arduino IDE upload tool. But i want upload .ino.bin via TCP/UDP server.
Can you tell me how to sort out this issue ?
 I did read one site it run on special UDP command and send hole file particular tcp port.
I do not set specific port,hostname and password.Is their any query than ask me?

d-a-v, 2017-04-10:
The utility is in tools/espota.py. It uses UDP and port 8266 to send binary flash file to your esp.
The server is your ESP.
The client is the utility.

Joypacharya, 2017-04-11:
i want to send file from TCP server,which developed in other device.
Is their any way to send this file ?
Give me exact flow.

d-a-v, 2017-04-14:
Maybe you'll find your answer there:
https://github.com/esp8266/Arduino/blob/master/doc/ota_updates/readme.md#http-server

devyte, 2017-09-30:
Closing due to off-topic, please see #3655 .

closed 2017-09-30 by devyte