esp8266/Arduino #3084: Additional boards manager url option is missing from arduino 2:1.0.5+dfsg2-4

soumen111, 2017-03-25:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

closed 2017-05-15 by igrr

shreve, 2017-05-23:
Why was this closed with no explanation? I'm having this issue as well.

igrr, 2017-05-24:
The OP didn't provide any description of the issue and submitted the placeholder intended to be filled in instead. If you are also having the issue and can provide some details what this is about, please feel free to do so :)