esp8266/Arduino #3085: Could not compile on clean Arduino IDE

kokss3, 2017-03-25:
Reinstalled library Arduino IDE and installed via Board Manager...

But compiler constantly asks for some new library ("queue.h", "functional", "bits/c++config.h", ...)
I have checked case, location and all other ways to resolve this, but still no success....

_____last problem:
C:\Users\xxxx\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiType.h:26:19: fatal error: queue.h: No such file or directory

 #include <queue.h>

______Solved and got other problem:
C:\Users\xxxx\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27:22: fatal error: functional: No such file or directory

 #include <functional>

______And another:
C:\Users\xxxx\Documents\Arduino\libraries\ESP8266WiFi\src/functional:34:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

How many more, and what is possible solution?

JamesGKent, 2017-03-27:
what operating system are you working on? if XP you might need to try the fix here:  
[https://github.com/esp8266/Arduino/issues/1631#issuecomment-192680550](https://github.com/esp8266/Arduino/issues/1631#issuecomment-192680550)

kokss3, 2017-03-27:
it is win7 x64

JamesGKent, 2017-03-27:
ok, can you post the complete error to start with, and also, you state that you "solved" the last error (first one listed above) how did you do this?

kokss3, 2017-03-27:
In file included from C:\Users\Koks\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27:0,

                 from C:\Users\Koks\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Users\Koks\Documents\Arduino\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\Users\Koks\Documents\Arduino\libraries\ESP8266WiFi\examples\WiFiAccessPoint\WiFiAccessPoint.ino:33:

C:\Users\Koks\Documents\Arduino\libraries\ESP8266WiFi\src/functional:34:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board Arduino Nano.

neu-rah, 2017-03-27:
you are compiling for a nano board, it should be some ESP
add ESP suppor if needed
https://www.arduino.cc/en/guide/cores

kokss3, 2017-03-27:
oooou.....right

Ok then it is solved....thank You all
I am closing

closed 2017-05-15 by igrr