esp8266/Arduino #3087: Problem with function WiFi.hostByName()

sembsa, 2017-03-25:
I have a problem with WiFi.hostByName when I get an address on a local network with a .local domain that shows me the error "Host: raspberrypi.local lookup error: -5!" When I'm looking for eg "google.com" it's all ok. If I'm doing pings on my computer the name raspberrypi.local is all ok. With every name on the local network is a problem: /

adriancuzman, 2017-08-22:
I think mDNS lookup functionality is not implemented in hostByName, but I found a library that is the missing part, mDNSResolver, specifically made to fix this issue :)
https://github.com/madpilot/mDNSResolver

stefaandesmet2003, 2018-02-19:
I have a similar issue with lwip2.
I modified the BasicHttpClient to connect to a non-existing .local address : 
http.begin("http://nonexistingname.local/"); //HTTP
and dns_gethostname returns the 192.168.1.8, the ip-address of my laptop (???), which of course refuses the connection.
When I disabled the laptop's wifi, the lookup fails with error -5 (that seems ok)

[BasicHttpClient+SerialOutput.zip](https://github.com/esp8266/Arduino/files/1738312/BasicHttpClient.SerialOutput.zip)

JaronrH, 2018-07-02:
Same issue here!

JaronrH, 2018-07-03:
Just confirmed that when I downgraded to v2.3.0 of the boards, DNS lookups worked flawlessly.

UnSafeHaven, 2018-09-10:
Real DNS lookup (Windows Server 2008 R2) of local addresses fails on version 2.4.2 when LwIP version 2 is used.  Switching to LwIP version 1.4 resolves the issue.

devyte, 2019-01-11:
@sembsa the first reply is correct: hostByName() does dns lookup. If you need to look up a .local, then you have to use the MDNS responder lib.
Closing.

closed 2019-01-11 by devyte