esp8266/Arduino #3089: Light Sleep Wakeup - Support Rising / Falling

SciLor, 2017-03-26:
### Basic Infos

#### Hardware
NodeMCU v1.0
### Description

The ESP8266 only wakes up when you use GPIO_PIN_INTR_LOLEVEL / GPIO_PIN_INTR_HILEVEL,
So for GPIO_PIN_INTR_LOLEVEL you would need to Pull the voltage down to 0V then to 3.3V up again.

I have an SIM808 Module. It has a ring indicator, which is normally HIGH and goes LOW when a call is comming in. It goes up again, when the call is rejected/accepted.
See: https://www.dfrobot.com/wiki/index.php/SIM808_with_Leonardo_mainboard_SKU:DFR0355#About_JP2

The power usage of the NodeMCU in Sleep Mode is around 17mA. When the Ring is LOW, it rises to 86mA. It seems the ESP8266 woke up. But the codeflow on the ESP8266 is only continued if I reject the call and Ring is HIGH again.

The solution would be a using GPIO_PIN_INTR_ANYEDGE or GPIO_PIN_INTR_NEGEDGE. But If I use this, nothing happens. So maybe something is broken? Or is the ESP8266 firmware itself missing the implementation of that options?

### Sketch

```cpp

	wifi_fpm_set_sleep_type(LIGHT_SLEEP_T); // set sleep type, the above posters wifi_set_sleep_type() didnt seem to work for me although it did let me compile and upload with no errors 

	gpio_pin_wakeup_enable(CONFIG_GSM_RING_PIN, GPIO_PIN_INTR_ANYEDGE); // GPIO_ID_PIN(2) corresponds to GPIO2 on ESP8266-01 , GPIO_PIN_INTR_LOLEVEL for a logic low, can also do other interrupts, see gpio.h above

	wifi_fpm_open(); // Enables force sleep
	wifi_fpm_do_sleep(0xFFFFFFF); // Sleep for longest possible time
```

### Debug Messages

```
messages here
```

soheils2, 2017-07-30:
hi there same issue here !
I search into some codes and libraries and found this low level function at "gpio.c" :

`
/*
  only highlevel and lowlevel intr can use for wakeup
*/
void ICACHE_FLASH_ATTR
gpio_pin_wakeup_enable(uint32 i, GPIO_INT_TYPE intr_state)
{
    uint32 pin_reg;

    if ((intr_state == GPIO_PIN_INTR_LOLEVEL) || (intr_state == GPIO_PIN_INTR_HILEVEL)) {
        portENTER_CRITICAL();

        pin_reg = GPIO_REG_READ(GPIO_PIN_ADDR(i));
        pin_reg &= (~GPIO_PIN_INT_TYPE_MASK);
        pin_reg |= (intr_state << GPIO_PIN_INT_TYPE_LSB);
        pin_reg |= GPIO_PIN_WAKEUP_ENABLE_SET(GPIO_WAKEUP_ENABLE);
        GPIO_REG_WRITE(GPIO_PIN_ADDR(i), pin_reg);

        portEXIT_CRITICAL();
    }
}

`

from this library [https://github.com/espressif/ESP8266_MP3_DECODER](url)
and this part kinda make me upset:
`
/*
  only highlevel and lowlevel intr can use for wakeup
*/

`
now my question is: is there any way to wake up my ESP guy from light sleep using falling or rising  edges?

actually
i trying to wake my esp by a gpio that defined as software serial.
and when the board on the other side is ON we always get HIGH (if highlevel int attached it goes on the bad loop) then i must chose low level
and would be the same when the board on the other side is OFF we always get LOW (if lowlevel int attached it goes on the bad loop)       then i must chose high level int

can anybody help me with that?

SciLor, 2017-07-30:
I found no solution yet.

devyte, 2017-07-30:
Hi,

Did you read the espressif NONOS sdk api reference, to figure out the
capabilities for wakup? If that supports edges, you can use the api
directly if this arduino core lib doesn't. If espressif doesn't support it,
then there's nothing you can do.

devyte, 2017-10-19:
@SciLor per Espressif's [document 9b](https://www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf) ony HIGH or LOW level can wake the ESP. EDGE wakeup is not supported.
Closing.

closed 2017-10-19 by devyte