esp8266/Arduino #3096: ESP8266WebServer.h and images from de SPIFFS (JPG files)

ESPlorador, 2017-03-28:
### Basic Infos

#### Hardware
NodeMCU

### Description

I´m trying to extract an image from the SPIFFS but I can´t. Does anyone what´s the problem???

### Settings in IDE

Flash Size: 		4MB

### Sketch
#include <ESP8266WiFi.h>                      
#include <ESP8266WebServer.h>                 
#include "FS.h"                               
 
const char *ssid = "IMAGE",                    
           *password = "12345678";               
String content;
String texto1;
String texto2;
 
ESP8266WebServer server(80);            
 
void setup() {
 
  Serial.begin(115200); Serial.println("");   
  SPIFFS.begin();                             
 
  WiFi.softAP(ssid, password);                
 
  server.on("/image", []() {
      File f = SPIFFS.open("/sonic.jpg", "r");
      server.streamFile(f, "image/jpeg");
      int filesize = f.size();
      String WebString = "";
      WebString += "HTTP/1.1 200 OK\r\n";
      WebString += "Content-Type: image/jpeg\r\n";
      WebString += "Content-Length: " + String(filesize) + "\r\n";
      WebString += "\r\n";
      server.send(200, "text/html", WebString);
      
      char buf[1024];
      int siz = f.size();
      while(siz > 0) {
        size_t len = std::min((int)(sizeof(buf) - 1), siz);
        f.read((uint8_t *)buf, len);
        //server.client().write((const char*)buf, len);
        server.sendContent((const char*)buf);
        siz -= len; 
        f.close();
      } 
  }); 

  server.begin();                            
}
 
void loop() {
  server.handleClient();
}
```

### Debug Messages

```
None
```

TalkThing, 2017-03-29:
qwe

jjssoftware, 2017-03-30:
This looks ok but I'm saying that by just reading your code alone - I'm not actually running / testing it. 

`server.streamFile(f, "image/jpeg");` should be returning the image data so long as the /sonic.jpg file is in the SPIFFS. 

Please describe in detail what is it that you see and experience when you run and test this.

ESPlorador, 2017-03-31:
When I write **http://192.168.4.1/sonic.jpg** in the browser the message is: **Not found: /sonic.jpg**
Of couse sonic.jpg is in the SPIFFS.

jjssoftware, 2017-04-01:
The code in your [comment #1](https://github.com/esp8266/Arduino/issues/3096#issue-217718753) is being edited so I guess what I'm commenting on is sort of "in flux". 

That said what I said still stands so far as `server.streamFile(f, "image/jpeg");` alone will be returning image data. 

I'm not surprised that this flavour of your code isn't working as expected and that a browser can't make sense of the response because your `server.on("/image"` handler is actually returning 3 separate responses which are all going to be joined together into one long invalid broken HTTP response which will be completely confusing the receiving client side. 

Developer tools network tab in a browser will show you your actual raw request / response.

Your single request handler now implements:

* `server.streamFile`
* `server.send`
* `server.sendContent`

Use of SPIFFS isn't the problem here. The way you've built this request handler code is the cause of this problem.

I suggest that you have a think about what type of response you want / need to deliver to the client, how you're expecting the client to be able to process this response and then construct a single response that contains code that delivers on this requirement. I guarantee that this will work.

ESPlorador, 2017-04-02:
Thaks. 
Now I know what I have to solve.

nbmachado, 2017-08-19:
Já tem um código que funcione?
compartilhe

Already have a code that works?
Share

devyte, 2017-09-07:
This is user error in the endpoint callback. Closing.

closed 2017-09-07 by devyte

PirSoft, 2019-09-18:
> When I write **http://192.168.4.1/image** in the browser recives only a small part of the image.
> 
> Internet Explorer doing click on the image to see de propertys tells that doesn't know the size or the kind of file.

how is your solution?  pls inform to me. thanks.

PirSoft, 2019-09-18:
i used below code.  only get small part image from sd card.
problem?   
thanks.
`
      File f = SD_MMC.open("/olgun.jpg", "r");
      int filesize = f.size();
      String WebString = "";
      WebString += "HTTP/1.1 200 OK\r\n";
      WebString += "Content-Type: image/jpeg\r\n";
      WebString += "Content-Length: " + String(filesize) + "\r\n";
      WebString += "\r\n";
      //server.sendContent(WebString);
      //server.send(200, "text/html", WebString);
      // httpd_resp_send(req, WebString, strlen(WebString));
      
      char buf[1024];
      int siz = f.size();
      while(siz > 0) {
        size_t len = std::min((int)(sizeof(buf) - 1), siz);
        f.read((uint8_t *)buf, len);
        httpd_resp_set_type(req, "image/jpeg");
        httpd_resp_send(req,(const char*)buf, len);
        //httpd_resp_send(req,(const char*)buf));
        siz -= len; 
        f.close();
      } `