esp8266/Arduino #3100: WiFi only works once after erasing flash

jcollie, 2017-03-31:
### Basic Infos

WiFi only works once after erasing flash using esptool and then uploading a sketch.

#### Hardware
Hardware:			Adafruit Huzzah Feather
Core Version:      	git from today

### Description

If I use esptool to erase the flash and then use the Arduino IDE to upload a sketch, WiFi connects and works fine. If I reset the ESP8266 or cycle the power, WiFi never works properly again. The WiFi can authenticate and get a DHCP address but TCP connections or DNS lookups don't work. Using a Ubiquity AP-Pro as my WiFi access point. The ESP8266 is the only WiFi client that I've had problems with this AP.

### Settings in IDE

Module:  			Adafruit HUZZAH ESP8266
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			QIO
Flash Frequency:	80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

// Adafruit IO Temperature & Humidity Example
// Tutorial Link: https://learn.adafruit.com/adafruit-io-basics-temperature-and-humidity
//
// Adafruit invests time and resources providing this open source code.
// Please support Adafruit and open source hardware by purchasing
// products from Adafruit!
//
// Written by Todd Treece for Adafruit Industries
// Copyright (c) 2016-2017 Adafruit Industries
// Licensed under the MIT license.
//
// All text above must be included in any redistribution.

/************************** Configuration ***********************************/

// edit the config.h tab and enter your Adafruit IO credentials
// and any additional configuration needed for WiFi, cellular,
// or ethernet clients.
#include "config.h"

/************************ Example Starts Here *******************************/
#include <Adafruit_Sensor.h>
#include <DHT.h>
#include <DHT_U.h>
#include <Adafruit_SSD1306.h>

// oled display
Adafruit_SSD1306 oled = Adafruit_SSD1306();

// pin connected to DH22 data line
#define DATA_PIN 2

// create DHT22 instance
DHT_Unified dht(DATA_PIN, DHT22);

// set up the 'temperature' and 'humidity' feeds

AdafruitIO_WiFi *io;
AdafruitIO_Feed *temperature;
AdafruitIO_Feed *humidity;

void setup() {
  // start the serial connection
  Serial.begin(115200);
[... 642 lines cut ...]
wifi evt: 7
[hostByName] Host: io.adafruit.com lookup error: -5!
Disconnected from Adafruit IO.
[hostByName] request IP for: io.adafruit.com
[hostByName] Host: io.adafruit.com lookup error: -5!
Disconnected from Adafruit IO.
[hostByName] request IP for: io.adafruit.com
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
[hostByName] Host: io.adafruit.com lookup error: -5!
Disconnected from Adafruit IO.
[hostByName] request IP for: io.adafruit.com
[hostByName] Host: io.adafruit.com lookup error: -5!
Disconnected from Adafruit IO.
[hostByName] request IP for: io.adafruit.com
[hostByName] Host: io.adafruit.com lookup error: -5!
Disconnected from Adafruit IO.
[hostByName] request IP for: io.adafruit.com

```

maxdd, 2017-04-12:
I have exactly the same problem. Did you find a workaround? I have a NodeMCU chip and even with the flasher it is not resetting the credentials.
Hence i can only connect to the first entered network.

jcollie, 2017-04-12:
Yes, if I always set the WiFi mode, like such:

```
WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
```

If I leave off the WiFi.mode call I get the behavior described above. A lot of the sample code that I found through Google doesn't have a call to WiFi.mode.

maxdd, 2017-04-12:
Can you link me which SDK/firmware and method you used to update your device? i feel like i have an old version.

jcollie, 2017-04-13:
I'm using 2.3.0, nothing too special.

https://github.com/esp8266/Arduino/releases/tag/2.3.0

cbrum11, 2017-04-14:
I agree with @jcollie.

I use the following with all my ESP modules and it fixed my re connection (or lacktherof) issues.

**At the top of code:**
```
extern "C"{
#include "user_interface.h"
}
```

**In my setup_wifi() function**

```
void setup_wifi() {

  delay(10);
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  wifi_set_sleep_type(NONE_SLEEP_T);
  WiFi.persistent(false);     //These 3 lines are a required work-around
  WiFi.mode(WIFI_OFF);    //otherwise the module will not reconnect
  WiFi.mode(WIFI_STA);    //if it gets disconnected
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
```
Not sure what kind of black magic happens, with this, but it always seems to work.

-Chase

maxdd, 2017-04-14:
WiFi.persistent(false) --> doesn't it avoid the saving of the SSID/password in the flash?
So this snippet works only if you are writing the credentials in the code.
We need to find a solution for when we use SoftAP fallback or WiFiManager.
For a project like mine, i don't know the credentials and i need to let the user input them

cbrum11, 2017-04-14:
From [HERE](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/doc/esp8266wifi/generic-class.md)

> ### persistent
> 38	
> 39	```cpp
> 40	WiFi.persistent (persistent) 
> 41	```
> 42	
> 43	Module is able to reconnect to last used Wi-Fi network on power up or reset basing on settings stored in specific sectors of flash memory. By default these settings are written to flash each time they are used in functions like `WiFi.begin(ssid, password)`. This happens no matter if SSID or password has been actually changed.
> …	
> 45	This might result in some wear of flash memory depending on how often such functions are called.
> 46	
> 47	Setting `persistent` to `false` will get SSID / password written to flash only if currently used values do not match what is already stored in flash.

So the way I read that is...  it doesn't avoid saving the SSID/password to flash, it avoids **REWRITING** the values to flash if they're not different than the last values used.

Either way, apologies, because I don't think I've been much help to your current dilemma.

maxdd, 2017-04-14:
That's interesting, maybe i confused it with wifi.disconnect(false) ?
Anyway i have two questions for you,
1. Why are you using wifi_set_sleep_type(NONE_SLEEP_T) in the setup() ?
2. Based on your experience how would you change this https://github.com/esp8266/Arduino/issues/3132 ? Would you put 
```
WiFi.persistent(false);     //These 3 lines are a required work-around
WiFi.mode(WIFI_OFF);    //otherwise the module will not reconnect
```
both before 

```
WiFi.mode(WIFI_STA);
WiFi.begin(WiFi.SSID().c_str(), WiFi.psk().c_str());
```

and 

```
WiFi.mode(WIFI_STA);
WiFi.begin(SSID.c_str(), PW.c_str());
```

devyte, 2019-11-10:
This issue is old. A lot has changed since it was reported, including several fixes that are relevant to the original post.
As to the light sleep post, that is still ongoing and is a known issue with our current SDK 2.2.x branch. I think the issue has to do with how sleep is done: there are two light sleep levels, min and max. In 2.2.x the level is hardcoded to max. In SDK 3.x there is an api that allows choosing min or max. Level max is appropriate for cases where the ESP wakes up and initiates outbound connections and then shuts down again. Level min is appropriate for server-type cases where somebody needs to connect to the ESP over network.
Given that 2.2.x doesn't have level min, the only reliable setting for server-type apps is no sleep.

Closing this.

closed 2019-11-10 by devyte