esp8266/Arduino #3104: Can Hazzah support my race timer application

harleyfrazee, 2017-04-01:
My application involves two "beam break" devices and an iPhone. Both beam break devices send UDP keepalive messages every 1 second to the iPhone and send a different USP message when the beam is broken as soon as possible since the desired timing would be in 100th of a second.

Due to iPhone security one of the Hazzah devices must be a softAP (the other a WiFi client). The softAP functionality must include:
   1) software programmable SSID/BSSID
   2) IP address of AP predictable and WiFi connected devices must be giving appropriate IP addresses
  3) after the devices are WiFi connected they should be able to send and receive UDP messages both broadcast and unicast. 
  4) Hardware/firmware wise, the device must be able to detect a rising or falling voltage signal on a pin in order to support a beam break.
   5) Hardware/firmware wise, it would be desired to be able to use one of the pins to turn on/off an LED. Even more desirable to support a dual color LED. 
  6) Hardware/firmware wise, it would be desired to be able to detect a low-battery signal.

tttapa, 2017-04-22:
1. What is a Hazzah?
2. UDP packets don't always arrive. It's not reliable. You could use WebSocket for low-latency, reliable communication. 
3. A wired solution would be much more accurate. (and easier)

Pieter

themindfactory, 2017-04-22:
These post are meant for bug report and upgrade information to the Arduino
source code and not for support questions that Adafruit should handle.

Too many people using this service inappropriately.

Have a question then the manufacturer or esp8266.com

RichardS

devyte, 2017-09-07:
This is not the right place to ask for support for your particular project. Please request assistance at esp8266.com or stackoveflow.
Closing.

commit 8ee35d0c (2015-06-26): Cut off board/port name at 50 instead of 16. Fixes #3104
closed 2017-09-07 by devyte