esp8266/Arduino #3108: WROOM-02 reset when client.print more "long" messages

mersasmith, 2017-04-03:
Hello,

I'm having a very annoying problem with this library. I'm trying to send a message (with client.print()) of about 70 bytes. Every time I try to send this message the module resets. The only way to send data without these resets is to send messages with a length lower than 25 bytes.

I saw in the documentation that this library not allow more than 25 bytes per message, but I don't understand why the module is being reset when this value is supased. 

There are any kind of method to allow larger messages?

Thanks for the support.

jjssoftware, 2017-04-12:
This reset sounds like a watchdog reset or maybe a power supply problem. I'm not sure where you saw a documented 25 byte limitation though - where is that documented?

devyte, 2017-09-07:
I know that client.print() currently works with messages longer than 25 bytes.
@mersasmith is this still an issue for you? please post a full minimal sketch that reproduces the issue, and additional info requested in the issue template. Also, please make sure to try with latest git.

devyte, 2018-02-28:
Closing due to lack of feedback and age.

closed 2018-02-28 by devyte