esp8266/Arduino #3109: Power Cycle vs Reset / CH_EN

vlast3k, 2017-04-03:
### Basic Infos

#### Hardware
Hardware:			ESP-12e
Core Version:      	2.3.0

### Description

While fighting with some issues i have with SPIFFS i noticed that in some occasions, calling SPIFFS.begin() results in a fatal exception (still debugging it), and then regardless if i do short RST pin or CH_PD pin, i get one and the same result

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0xad
csum 0xad
v40f0c112
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
epc1=0x400ef424, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
Fatal exception (0): 
.....
```

but if i powercycle the ESP, then my sketch loads again, and executes again until the place where i call SPIFFS.begin().

So could somebody elaborate what may be the difference between doing a complete power cycle and simply resetting the board via the pins ?

devyte, 2017-09-07:
I'm not an expert in this, but I believe that SPIFFS.begin() mounts the filesystem, during which it checks for certain magic bytes. When it fails, it is generally an indication that there is something wrong with either the SPIFFS image itself, or the flash layout.
How are you making the spiffs image? if you are not using an image, did you format it on first boot?
If you are using an image, how are you flashing it?

devyte, 2017-10-10:
No sketch, no feedback in over a month. I don't see issues with SPIFFS.begin in my own app.
Closing.

closed 2017-10-10 by devyte