esp8266/Arduino #3110: ESP12 throwing exception 29 + AP mode + HTTP

anujmattoo, 2017-04-03:
#### Hardware

Hardware:			ESP-12F
Core Version:      	        2.3.0

### Description

I am working on RGB led controller and controlling the lights from mobile app and sending the HTTP commands to ESP in AP mode.

The code works perfectly initially and when using few commands and when speed up the HTTP commands from Mobile app to ESP, the ESP works for sometime and then crashes and reset.
I am sending around 20-30 request per second.

I am using git core version of ESP8266 library for Arduino.

How to solve this issue, earlier it was solved by using the git core version, but it is crashing again. What to do?

### Settings in IDE

Module:  			Node MCU ESP12E
Flash Size: 		        4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL

### Sketch

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

#define RED_LED     12
#define WHITE_LED   13
#define BLUE_LED    14 //16  //14
#define GREEN_LED   16 //14  //15

char ssid[10] = "Downlight";

WiFiServer server(5045);
WiFiClient client;

void setup()
{
  pinMode(RED_LED, OUTPUT);
  pinMode(WHITE_LED, OUTPUT);
  pinMode(BLUE_LED, OUTPUT);
  pinMode(GREEN_LED, OUTPUT);

  WiFi.mode(WIFI_AP);

  Serial.begin(115200); //Start communication between the ESP8266-12E and the monitor window
  WiFi.softAP(ssid);

  server.begin();
}

void loop()
{
  client = server.available();
  if (!client)
  {
    return;
  }
  
  String req = client.readStringUntil('\r');
  Serial1.println(req);
  client.flush();

  if ((req.indexOf("/RGB/") != -1))
  {
    int pos = req.length();
    int ind1 = req.indexOf("/RGB/") + 5;
    String teststring = req.substring(ind1, pos);

    int ind2 = teststring.indexOf("/");
    String hexRGB = req.substring(ind1, ind2 + ind1);

    Serial.print("\n");
    Serial.print("Hex RGB: ");
    Serial.print(hexRGB);

    hexRGB.toUpperCase();
    char c[6];

    hexRGB.toCharArray(c, 7);
    long r = convertToInt(c[0], c[1]); //red
    long g = convertToInt(c[2], c[3]); //green
    long b = convertToInt(c[4], c[5]); //blue

    analogWrite(RED_LED, r * 4);
    analogWrite(GREEN_LED, g * 4);
    analogWrite(BLUE_LED, b * 4);

    client.flush();
  }
}
int convertToInt(char upper, char lower)
{
  int uVal = (int)upper;
  int lVal = (int)lower;
  uVal = uVal > 64 ? uVal - 55 : uVal - 48;
  uVal = uVal << 4;
  lVal = lVal > 64 ? lVal - 55 : lVal - 48;
  return uVal + lVal;
} 
```

### Debug Messages

Exception Decoder result:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

StefanFrings, 2017-04-15:
I don't see any error in your source code. But I have a very similar problem, therefore I entered your thread.
I am not able to run a Network performance test because my ESP-12F crashes after a few seconds when I send lots of data to it.

The is my minimal code which can be used to reproduce the issue:
`#include <ESP8266WiFi.h>

#define SSID "Muschikatze"
#define PASSWORD "supergeheim"
#define PORT 5333

WiFiServer server(PORT);

void setup()
{
    WiFi.mode(WIFI_AP_STA);
    WiFi.begin(SSID, PASSWORD);
    server.begin();
}

void loop()
{
    WiFiClient client = server.available();
    if (client)
    {
        int avail = client->available();
        if (avail)
        {
            char buffer[avail + 1];
            clients[i]->readBytes(buffer, avail);
            buffer[avail] = 0;
            // Place code to process the received data here
        }
    }    
}`

My ESP-12F module has an additional 100µF capaciator solderect directly on top of it. RST and CH_PD are hardwired to VCC. GPIO15 has a 2,2k Pulldown resistor. I'm sure that the power supply is not the problem cause, I tried different devices with linear 3,3V regulator as well as 3xAA NiMh batteries.

When I use netcat to send data manually, I have not problem. But when I send much data (10kB or more) with little or no delay, then the ESP crashes after a random time - but always withing one minute.

Example killer command for the Cygwin shell:
(sleep 5; cat largefile.txt) | nc6 192.168.2.40 5333

Using the --disable-nagle and/or --mtu=100 options did not help.

On the other hand, my ESP can send a lot of data continuously to my Laptop. The performance looks pretty fine for the sending direction: 200 - 300 kByte/sec.

But when the ESP receives more than 8 packages of 1kByte per seconds, then it crahes.

Any Ideas how to fix it?

StefanFrings, 2017-04-15:
While working on a bulletproof problem description for you, I found the solution myself :-)
It seems that sometimes much more than 1kByte become available which leads to a stack overflow.
I changed the code now so it does not read more than 1kB in each loop iteration. Now it works stable and pretty fast.

```
if (avail)
                {
                    if (avail>1024)
                    {
                        avail=1024;
                    }
                    char buffer[avail + 1];
                    client.readBytes(buffer, avail);
                    buffer[avail] = 0;
                    sum+=avail;
                    // Place code to process the received data here
                }
```

StefanFrings, 2017-04-15:
@anujmattoo:
Looking back to your code, I see now one line that may be affected by the same issue as my program:
```
String req = client.readStringUntil('\r');
```
Imagine what happens if that line is more 10kB large. Then you have basically the same problem - a stack overflow.

devyte, 2017-09-08:
@StefanFrings thank you for that solution, but String doesn't allocate on the stack. It holds internally a managed pointer to a chunk on the heap.

@anujmattoo 20-30 requests per second is pushing it. Most likely, the ESP can't finish processing one request before the next one is already in. 30 requests per second means about 33ms time to receive and handle the request, without counting the time needed to service the wifi stack.
In addition, once a connection is finished, the socket is usually left in a TIME_WAIT state for a short period of time. The espconn layer tries to get around this by offering only a limited number of sockers, and resetting and reusing those that are in TIME_WAIT state, but from conversations I have read, making use of that is not very stable.
You are essentially doing a type of DoS attack, and due to its limited resources, the ESP can't handle that kind of thing.
I consider this a usage error, so closing.

closed 2017-09-08 by devyte

StefanFrings, 2017-09-08:
I'm not satisfied by this answer. 

It does not matter where Strings are stored. They can easily overflow the available memory.  But I understand that nobody will change the behaviour of the String class. Man should rather improve the programming examples. Explain why reading data of unknown length is dangerous in general. The very basic beginners example should already show how to do it right (maybe as I did above).

The other part of the answer "You are essentially doing a type of DoS attack" is not acceptable. Any type of computer is not allowed to crash when it get fully loaded. I know for sure that the underlying LwIp stack ha no problems with high load as well as with lots of connections. Even the older µIP stack has no such problem. But anyway, if there is another limitation outside the IP stack, then the firmware should simply reject or delay incoming connections. When the firmware accepts a connection then it must be able to handle it. Every other behaviour is broken by design.

Makuna, 2017-09-08:
I want to chime in here with something I have learned.

DO NOT use the String class.  Period.  It is poorly designed to handle common errors (out of memory is the most common) and is aimed at the most novice developers.  It was something designed way back in the early days of Arduino and really needs to reworked.

I wish the Wifi classes just didn't use it.  But at best the WiFi classes should provide similar features without the need to use the String class.

devyte, 2017-09-08:
@StefanFrings 

> It does not matter where Strings are stored

Your previous comment was talking about a stack overflow. The ESP stack is limited to about 4KB, so if you declare a char array of variable size on the stack, and that size is beyond the remaining available stack size, the stack will blow up. In that sense, your first assessment was correct, and your fix of limiting to 1KB at a time was good.
However, a String object does not hold its content on the stack, but rather on the heap, and is therefore limited only to the biggest available heap chunk. Yes, you could still "overflow memory" if you receive a huge chunk, say 20KB in size, thereby using most or all available mem on the ESP.
My comment points out that using String is a different issue than blowing up the stack.

About overwhelming the ESP: If this core were built solely on lwip, we might be able to make it robust against out of resource situations. However, there is the espressif SDK in between, which has a long list of issues, instabilities, bad error checking, etc, and we don't have full access to that code. So yes, there is a limitation outside of the lwip stack, which is a not-so-well-written code layer.
It is possible to eliminate some or most of the espressif communications layer. The Nodemcu project did that to a large extent. It did help in making things more robust. but there are things to worry about, such as TIME_WAIT sockets, which can still eat up resources.
If you have experience with lwip, or IP stacks in general, please feel free to investigate and propose a solution. We're open to contributions, especially those that help improve overall stability.

devyte, 2017-09-08:
@Makuna 
The problem with any object that manages resources is how do you notify the user when you are faced with a failure. In traditional C you can check for a null ptr in user code. In traditional C++ you throw and catch. Here on the ESP, we don't have access to the underlying pointer, and we don't have exceptions (it would blow up mem and flash usage*). Given that, if a String needs to reallocate, and the reallocate fails, how do you let the user know of that failure?
The same is true of pretty much all std containers at this point. Consider std::vector: if you push_back and it triggers a realloc, and that realloc fails, the expected behavior would be to throw. But we don't have exceptions, so.....

*In my time here, I haven't seen any thorough investigation with hard numbers about this, meaning I haven't seen anyone enable exceptions in the compiler and measure the impact. I could be wrong, of course, or it could have been before I joined the repo way back when. But right now, as far as I know, this is theory. If somebody were to investigate the impact of enabling exceptions, and it turned out to be acceptable, we could have a whole lot of code improvements with try-throw-catch scenarios, including a definitive fix for String and fully operational std containers.