esp8266/Arduino #3114: HTTP Client.Begin Crashing with Variable

TheCamoc, 2017-04-04:
When i use a variable in http.begin the ESP resets.
 ets Jan  8 2013,rst cause:2, boot mode:(1,0)

 ets Jan  8 2013,rst cause:4, boot mode:(1,0)

wdt reset

# My code: 
#include <Base64.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
#include <OneWire.h>
#include <DallasTemperature.h>

#define mySSID "------------"
#define PASS "----------------
#define OWire 2
OneWire oneWire (OWire);
DallasTemperature DS18B20(&oneWire);
String middleware = "192.168.88.11/middleware.php/data/c86897c0-195e-11e7-ac6a-d7b175cea660.json?operation=add&value=";
void setup() {
  Serial.begin(115200);
  ets_intr_lock(); // all intruptt off
  WiFi.begin(mySSID, PASS);
  while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
  }

}

void loop() {
  DS18B20.requestTemperatures();
  int temp = DS18B20.getTempCByIndex(0);
  Serial.print("Temperature: ");
  Serial.println(temp);
  String website =  middleware + temp;
  if (WiFi.status() == WL_CONNECTED) {
    HTTPClient http;  //Declare an object of class HTTPClient
    http.begin(website);  //Specify request destination
    int httpCode = http.GET();                                                                  //Send the request
    if (httpCode > 0) { //Check the returning code
      String payload = http.getString();   //Get the request response payload
      Serial.println(payload);                     //Print the response payload
    }
    http.end();
  }
}

When i replace the website with the String it works fine.

jjssoftware, 2017-04-05:
what type of ESP board is this? how are you powering your board?

TheCamoc, 2017-04-05:
Had it powered with my PC USB but after i connected it to an better supply it worked thank you.

closed 2017-04-05 by TheCamoc

jjssoftware, 2017-04-05:
Cool, it had the hallmarks of a brown out / poor power supply