esp8266/Arduino #3117: SCLK, MOSI, MISO, CS and GPIO9 and GPIO10 ?

bpham817, 2017-04-05:
Hi everyone,
I have a few questions about ESP8266-12E dev kit. Hopefully, anyone could shed some lights to my following questions:

1. Based on Arduino IDE 2.3.0 it looks like the pin outs of GPIO9 and GPIO10 are not supported in this release. Has anyone tried to configure and make those 2 pins as digital input yet?

1. What is the difference between SCLK, MOSI, MISO, CS and HSCLK, HMOSI, HMISO, HCS? how to use the SCLK, MOSI, MISO, CS pins? what level of changes in firmware that I need to change

Thanks all
Bp

yorik, 2017-04-05:
AFAIK, SCLK, MOSI, MISO, CS are attached to internal flash and should not be used for anything else (I tried to use them and failed). Also HCS which is GPIO10 is also problematic because if it's high during boot then the board will go into special boot mode and doesn't boot from flash.
It's a problem for me because BME280 makes it high.

bpham817, 2017-04-06:
thanks yorik

Assuming MOSI of GPIO13 is available to use, have you tried to make MOSI  as a Slave Select so I can add and control a 2nd sensor to my system? how do i make the changes in the SPI library?

Sg4Dylan, 2017-04-07:
Have you tried [this library](https://github.com/esp8266/Arduino/tree/master/libraries/SPISlave) ?

closed 2017-05-10 by bpham817