esp8266/Arduino #3120: Way to test 'newlib' on Arduino IDE for ESP8266

electronicsguy, 2017-04-06:
Hello,
      As mentioned here: [newlib](https://github.com/esp8266/Arduino/pull/1752), the libc for newlib was merged sometime back in 2016. 

    However, I am unable to use "test_newlib" located here: [test_newlib](https://github.com/esp8266/Arduino/tree/master/tests/device/test_newlib). It still gives me the error:

```
C:\...\AppData\Local\Temp\arduino_build_956035\sketch\test_newlib.ino.cpp.o: In function `TEST_FUNC__27':

C:\...\test_newlib/test_newlib.ino:30: undefined reference to `sscanf'
```
How do we use the newlib and in particular sscanf() within Arduino IDE for esp8266?

jjssoftware, 2017-04-06:
https://github.com/esp8266/Arduino/commit/d7d98d03cad80be3a21581a160ef8d22c219724b was committed just after the last official 2.3 release. If you want this you'll need to use git code.

devyte, 2017-09-07:
@electronicsguy were you able to use sscanf with latest git?

HarmOtten, 2017-12-05:
I tested sscanf with v2.4.0-rc2 and it works.
Thanks a lot!

closed 2017-12-05 by devyte