esp8266/Arduino #3124: ESP12E Uploading problem---->esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>

sarwadenj, 2017-04-08:
#### Hardware
Hardware:      --ESP-12e (uploading code in ESP12E)
### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		4MB/1MB    (what it should be?)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz?
Upload Using:		--->SERIAL
Reset Method:		--->ck 
------------------------------------------------------------------------------------------------------------------
C:\Users\vishal\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.9/esptool.exe -vv -cd ck -cb 115200 -cp COM3 -ca 0x00000 -cf C:\Users\vishal\AppData\Local\Temp\arduino_build_1597/ESPdroneFinalest.ino.bin 
esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM3
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1

My connection are
http://www.instructables.com/id/Getting-Started-with-the-ESP8266-ESP-12/?ALLSTEPS

GPIO 0--GND (WHILE UPLOADING)
GPIO 15-->GND
GPIO 2-->VCC

jjssoftware, 2017-04-09:
imo there's only one good way to answer this because no-one could possibly offer any more info than what's been documented here: 

https://github.com/esp8266/Arduino/blob/209c87715768b0f280bb85b3832883487dc4d2e9/doc/faq/a01-espcomm_sync-failed.md

jaromanda, 2017-04-18:
"GPIO 0--GND (WHILE UPLOADING)" - no need ... you only need GPIO0--GND while booting to get the ESP into flashing mode - i.e `ch_pd->gnd` **and** `gpio0->gnd` - hold `gpio0` at `GND` until you set `cp_pd->vcc`

devyte, 2018-02-28:
Closing due to age and documented.

closed 2018-02-28 by devyte