esp8266/Arduino #3126: Arduino 1.82 ESP8266

Hokahonay, 2017-04-08:
Hi
after reading the issue #824 on installing the latest core for ESP8266
I uninstalled Arduino and installed 1.8.2 following the instructions at https://github.com/esp8266/Arduino#using-git-version
after updating boards (installing 8266 core) I see on my system 
AppData\Local\Arduino15\packages\esp8266\hardware\esp8266 that the version of the ESP8266 core is still 2.3.0 as it was before uninstall
if I try to verify APEventHandler.ino I get the error:
error: 'class ESP8266WiFiClass' has no member named 'onSoftAPModeProbeRequestReceived'
have I missed something in the readme or made some other error when installing
thanks for any pointers
Dave

#### Hardware
Hardware:		ESP8266-01
Core Version:      	2.3.0
### Description

Problem description

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		1MB
CPU Frequency:	80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

Hokahonay, 2017-04-08:
Update, there are two handlers that will not compile:
  probeRequestPrintHandler = WiFi.onSoftAPModeProbeRequestReceived(&onProbeRequestPrint);
  probeRequestBlinkHandler = WiFi.onSoftAPModeProbeRequestReceived(&onProbeRequestBlink);
so its the WiFi.onSoftAPModeProbeRequestReceived which Arduino is complaining about

Hokahonay, 2017-04-08:
sorry, didn't mean to close, just not very good at using GitHub yet

stray77, 2017-04-10:
I believe you require 4MB flash.  I only have experience with a NodeMCU V1.0 which uses an ESP8266-12E and has 4MB.  OTA works great with a reliable wifi connection.

devyte, 2017-09-06:
@Hokahonay 
I think you have to remove the 2.3.0 directory, otherwise Arduino ignores the git. Try moving 
AppData\Local\Arduino15\packages\esp8266
outside of Arduino15 (save it e.g.: as AppData\Local\esp8266.2.3.0.bak or something, as backup).
Then restart the IDE.
If this works, please report back here, because the docs should be updated.

macjonesnz, 2017-09-13:
Add the staging version of the board and it will compile. (2.3.0 rc2)
http://arduino.esp8266.com/staging/package_esp8266com_index.json

asierfd, 2017-09-19:
I have same problem, I've tried removing 2.3.0 directory and reload again, doesn´t work
I've tried with 2.3.0.rc2 version and it tell me that it doesn´t know this device

devyte, 2018-02-28:
Closing as git installation is documented (I followed them myself first time).

closed 2018-02-28 by devyte