esp8266/Arduino #3127: Wire library slave examples won't compile with Arduino 101

Pachicito, 2017-04-10:
Works great with the Uno, but not the 101.

Slave_receiver example:
error: 'class TwoWire' has no member named 'onReceive'

Same with 'onRequest' on Slave_sender example (full error below)

Same issue in both desktop IDE and Create. Multiple people experiencing the same issue: https://forum.arduino.cc/index.php?topic=375503.0

Any help is much appreciated!

arduino-builder/arduino-builder -compile -core-api-version 10611 -build-path /tmp/304782680 -hardware arduino-builder/hardware -hardware arduino-builder/packages/cores -tools arduino-builder/tools -tools arduino-builder/packages/tools -built-in-libraries arduino-builder/latest -libraries /tmp/792166045/pinned -libraries /tmp/792166045/custom -fqbn Intel:arc32:arduino_101 -build-cache /tmp -verbose=false /tmp/792166045/slave_sender

/tmp/792166045/slave_sender/slave_sender.ino: In function ‘void setup()’:

/tmp/792166045/slave_sender/slave_sender.ino:18:8: error: ‘class TwoWire’ has no member named ‘onRequest’

Wire.onRequest(requestEvent); // register event

^

exit status 1

Pachicito, 2017-04-10:
The version of the Wire library I downloaded directly from Arduino Create says this in the library.properties file:  Specific implementation for Arduino Zero

ccooper21, 2017-04-13:
@Pachicito You have posted your issue in the wrong place.  This repository is for the Espressif ESP8266 core for use with the Arduino IDE.  The Arduino 101 board is based on an Intel provided core and not an Espressif core.  It looks like the proper place to get support for the Arduino 101 board is [here](https://forum.arduino.cc/index.php?board=103).

closed 2017-04-29 by igrr