esp8266/Arduino #3130: Is there a way for me to use the CountLeadingZero (CLZ) assembly instruction in a sketch?

mark-hahn, 2017-04-11:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

jjssoftware, 2017-04-12:
I don't have an exact answer about CLZ but there are many examples of assembler code in this project e.g. https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/cores/esp8266/Arduino.h#L118

So "accessing" assembler is basically simply a case of declaring a code construct with `asm` or `asm volatile` (according to the [gcc docs](http://gcc.gnu.org/onlinedocs/gcc-4.7.2/gcc/Extended-Asm.html), the volatile keyword indicates that the instruction has important side-effects). 

If the next question was "how do I construct the correct asm CLZ instruction for an ESP board?", this would be a very hardware specific question and since ESP8266 devices are based on the LX106 processor along with the Xtensa instruction set architecture, to answer that kind of question we'd need detailed reference documentation for that: https://www.google.com/search?q=Tensilica+Xtensa+Instruction+Set+Architecture&ie=utf-8&oe=utf-8

I'm sorry I don't have an exact CLZ answer, someone else might. But in the absence of anything else, this is a starting point.

igrr, 2017-04-12:
You may consider using GCC's `__builitin_clz()` intrinsic. It is implemented using the correct instructions on each architecture. For Xtensa, it uses `nsau` instruction.

mark-hahn, 2017-04-12:
Thanks to both of you.  I have plenty of info to dig into this.  Sounds like fun.

closed 2017-04-12 by mark-hahn