esp8266/Arduino #3131: How to SPIFFS updating existing file content, i.e. overwrite a line?

faultylee, 2017-04-11:
### Basic Infos
#### Hardware
Hardware: ESPMOD
Core Version: 2.2

### Description
I have a 16k file which is used as a lookup file. It works perfectly using SPIFFS. But I do need to update it's content from time to time, i.e. replacing a line or a single char. The size of the file will be fixed so there should be less complexity.

I noticed that when opening file, `w` or `w+` will truncate the file, I don't see any mode that will allow me to seek to a position and update from there, unless I missed out something. So the alternative will be to duplicate the file for every update? Its also acceptable for my use case, since updating the file is not that frequent, probably once or twice a month. If I were to go with this approach, can I actually open 2 files and copy it's content over? Will the new file stays in memory or directly into SPIFFS before `close()`?

Any better suggestion or workaround is most welcome. Thanks in advance.

### Settings in IDE

Module: Wemos D1 Mini / NodeMCU 1.0
Flash Size: 4MB(SPIFFS: 1MB)
CPU Frequency: 80Mhz
Flash Mode: qio
Flash Frequency: 40Mhz
Upload Using:	SERIAL
Reset Method:	 ck & nodemcu

jjssoftware, 2017-04-13:
sounds like a use case covered by https://bblanchon.github.io/ArduinoJson/

faultylee, 2017-04-13:
@jjssoftware thanks, the allocation for `ArduinoJson` is in the RAM, with my 16kb binary data, it's most likely needing more than that to hold it in `JSON` format. Please correct me if I'm wrong.

jjssoftware, 2017-04-13:
Ram yes but there are options: 

https://bblanchon.github.io/ArduinoJson/faq/what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer/

For 16k of data you'd likely opt for the DynamicJsonBuffer approach

faultylee, 2017-04-16:
thanks, I'll try it out. Btw, not sure if anyone has the same need, maybe an option to overwrite specific position in the file would be useful

faultylee, 2017-04-19:
@Bendito999 thank you, that's exactly what I was looking for, didn't expect it to be under `r+`. I keep reading the `w` and `a` option, never thought `r+` would means writing as well, lol. Yes, it works for me, I have an index for the binary file lookup, each "line" is 16 bytes, no `\r` since I have the index to locate it. The file is sized with padding to accommodate additional data in the future, so the final size is fixed as well.

Thanks a lot!!!

closed 2017-04-19 by faultylee

wchpikus, 2018-12-20:
Hi.
Did you know how to search some string in spiffs file?
I had calllog, and have to find number on this file according data behand and forward  this string..

wchpikus, 2018-12-27:
Hi
I can read this file,but searching works only before read all file to ram..better will be read one line, campare and if no match, read next.
If you have some solution, please;)

faultylee, 2018-12-27:
@wchpikus for me I'm using `fs.read()` and reading a fixed number of bytes every time. I keep the text all at the same length, and just check the specific bytes location to match what I need. 

In a totally different project which I have variable line length, I read a fixed number of bytes and look for `\n`, then copy the remaining bytes into a temp variable and read and append, then look for `\n` again repeatedly, at the same time checking for match.

I don't have the code for variable line length for esp8266 specifically.

wchpikus, 2018-12-27:
fs.read() return something?
Could you send me some example on priv?
How do you know where read next data from?

faultylee, 2018-12-27:
```c
File fs = SPIFFS.open("log", "r");
char buf[52];
while (fs.available()) {
    fs.read((uint8_t *)buf, 49); // read 49 bytes into buf
    if (buf[48] == '\n') {
    } else {
        buf[49] = fs.read();
        if (buf[49] == '\n') {
            buf[49] = 0;
        } else {
            buf[50] = fs.read();
            buf[50] = 0;
        }
    }
}
fs.close();
```
Note: this is the esp8266 fixed line length example, not the variable length once (although it might looked like one)
The lines are at least 49 bytes, if the last byte is not `\n`, I call `fs.read()` again to read one byte, and check. My max line length is only 51, so the code works without too much overhead.

`fs.read(buf, size)` is the typical read call for this type of function (look at c/c++ string manipulation or file read functions) which read a fixed number of bytes into the buffer. The function also returns the number of bytes read, and you should check it in case it's at the end of the file and it return less than what you asked for.

wchpikus, 2018-12-27:
Hi
Ok i see, but this loop read data from 0 position to \n sign.
On next call, this loop wrire again from 0?

I need read some part of data ( it can be end of \n and campare) and back to the same place of file +1.
Maybe i can not see it in your code.

wchpikus, 2018-12-27:
Hi
But actually i need to return to last readed char..this is what i looking for.
Do you try seek function?
It can read from x to y, but i do not understand how to implement it..

wchpikus, 2018-12-27:
https://github.com/wchpikus/spiffs/blob/master/spiffs.cpp
Function String read_contact_campare(String file_name,String callerid_number){ };
Function read file txt and looking for callerid_number.